supermerill/SuperSlicer #4404: crash

rbreeden1, 2024-08-17:
### Version
2.5.59

### Operating system type + version
Windows 11

### Behavior
* Crash during slicing
* Slice Image
* Slice Preview
* Crash, no information provided
  * _Screenshots if useful_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_
[egg.zip](https://github.com/user-attachments/files/16642955/egg.zip)

Gray-Stone, 2024-08-17:
same on linux, I tried to run it from terminal, but didn't see any log during crash, and I can't get a meaningful backtrace either.

Dougwar, 2024-08-28:
You are using a beta version, in the discripiton its say;

> #known issues:
> the title bar still show "2.5.59", it will be fixed in next version. The good version is in help->about.
> **It can crash when it's trying to create seam positions. fixed in nightly.**

Try using the real 2.5.59 version, it is the lastest stable version

rbreeden1, 2024-08-31:
Tried on 2.5.60 and had the same issue.

supermerill, 2024-09-04:
can't reproduce on my win10 with 2.5.59.13
[image]
I'll try on linux when i have time.