supermerill/SuperSlicer #4407: 2.5.60 crashes constantly when trying to slice a ring

mslmarc, 2024-08-19:
### Version
2.5.60

### Operating system type + version
windows 11 home

### Behavior
when trying to slice a ring (about 6 inches in diameter) the system gets to generating then crashes. tried on 3 different machines and filament types but results are the same
select file, select tpu and slice.
file to slice and be ready for uplaod
crashes and exits.

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/
[seal error.zip](https://github.com/user-attachments/files/16658414/seal.error.zip)

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_
[image]

cdedwards, 2024-08-20:
I find SS crashing all the time when slicing as it's generating gcode on some items.

Dougwar, 2024-08-28:
You are using a beta version, in the discripiton its say;
#known issues:
> the title bar still show "2.5.59", it will be fixed in next version. The good version is in help->about.
> **It can crash when it's trying to create seam positions. fixed in nightly.**

Try using 2.5.59 version, it is the lastest stable version

mslmarc, 2024-08-28:
Thank you. I'll take a look. Normally I download the most recent stable version.