supermerill/SuperSlicer #4411: Starting Gcode ignores M106 command under certain conditions

Rattiputz, 2024-08-30:
### What happened?

**Behaviour**

The "Start G-code" ignores M106 when expressed as a decimal number between 0 and 1:
[image]
Since this doesn't concern the cooling or toolhead fans, I cannot work around this bug in the filament cooling menu.

This does not happen with values large enough (20 for example):
[image]
I guess this avoids PWM values that are too low for a fan to spin (0-255).

I am aware that this has been an issue before (https://github.com/supermerill/SuperSlicer/issues/834#issue-774962065), but I presume the fix was only applied to numbers between 0 and 255.
A suitable fix would be to allow the number range between 0 and 1 (on a Duet 6hc running RepRap Firmware this command is recognized) or prevent the user from exporting the gcode as long as the argument is not in the correct range.

### Project file & How to reproduce

1. Enter a M106 command into the custom gcode section with S#.## as a decimal number between 0 and 1 (for example M106 P2 S0.7)
2. Export the gcode and skip to the custom gcode section
3. The M106 command will not be there

### Version

2.5.59

### Operating system

Windows 10

### Printer model

Voron 2.4r2 with Duet 6hc+3hc running RepRap Firmware

shcad, 2024-08-31:
Its somehow expected behaviour. According to the standard the M106 gcode the S parameter number has to be an integer from 0 to 255 (0 for 0% speed and 255 for 100%). So what is Your goal? To set to some percentage of full speed (like 0.2 which represents 20% of full speed)?

Rattiputz, 2024-08-31:
[The duet documentation](https://docs.duet3d.com/User_manual/Reference/Gcodes#m106-fan-on) states that values between 0.0 and 1.0 can be used, as well as 0-255.

[image]

My goal is to use the first interval since it is much more intuitive (70% is S0.7 or S179, 15% is S0.15 or S38, you chose which fits you the best).
Since this is supported at least on Duet boards (`M106 P2 S0.7` sets fan 2 to 70%) it would be best that SS didn't ignore the parameter completely if the chosen interval is 0.0-0.1 (without any warning your part cooling fan could be off for the entire print, for example).

shcad, 2024-08-31:
Duet support means Reprap type gcode extension. I still use SS 2.4.58.5 and this version export custom start gcode with M106 S0.2 (I have checked with gcode flavours set to Reprap,   Klipper and Marlin). So ir is (was) possible.

So I think we have wait for Meril to make it clear (bug or intentional change).

Have You set gcode flavour to Reprap in Your SS version?

Rattiputz, 2024-08-31:
I came to the same conclusion, my custom gcode was from way earlier when the fans still started. I guess somehow this snuck its way into SS again in the meantime. I also checked beforehand that RepRap Firmware flavor was indeed selected.

I hope this can help people that are now having the same issue!