supermerill/SuperSlicer #4414: overlaping between perimeters and infill lines

Tinchus2009, 2024-09-08:
### Version
2.5.60
### Operating system type + version
win11
### Behavior
I think I said it a lot of times but it worth doing it agai: my choice of superslicer is because I have always though is the slicer that gives us the most powerfull amount of config options, and so we can do great things at the slicing time.
This would be one of those times: im trying to print a fabric mesh like print on tpu, my objective is done but I have a small problem: using modifiers i print some areas with perimeters and some with no perimeters at all, just infill. But the areas where infill and perimeters meet, are somehow weak because of a weack connection. A screenshot show one of these areas. I have not found, if there is one, a parameter to make primeter and infill paths to overlap more. Is there such parameter?

_Is this a new feature request?_
may be? If such parameter described above doesnt exist, I think it would be usefull to have the abilitily to control overlapping of 2 paths, despite is they are infill or perimeter, in order to creat stronger bondings, specially on those areas where you use modifiers.

[image]

neophyl, 2024-09-10:
I think you are looking for "Infill/perimeter encroachment".  Its an expert level setting so you must be in that mode to see it.  
[image]

Located in Print Settings>Width & Flow>Overlap.  All the overlap settings are Red/Expert.

Tinchus2009, 2024-09-10:
I tried that, did not print it, just tried the config and looked at the sliced simulation. I dont see a change on the slice simulation iamge, even putting a value of 50%, Could be true that slice simulation wont show it?

neophyl, 2024-09-11:
This is with 2.5.59

10% Encroachment
[image]

50% Encroachment
[image]

Try it with something simple like a cube and a basic infill with 60.  Who knows, its always possible it could be broken in that version.

Nbeknel, 2024-09-11:
If I understand correctly, you have model consisting of two types of areas, one with perimeters and one without. Let's call the one with perimeters A and the one without B. You're trying to improve the improve the connection between the external perimeter from A with the infill from B, right?

One idea that comes to mind would be to increase the flow when printing external perimeters, for this go to the "Custom G-code" tab in the printer settings and find the "Between extrusion role change G-code" section. Add the following:
```
{if extrusion_role == "ExternalPerimeter"}
M221 S150 ; increase flow
{else}
M221 S100 ; reset flow
{endif}
```
(150 is an example value, increase or decrease it to your needs).
With this you might want to set the `external_perimeter_overlap` value to 0.
An alternative way to increase the flow of the external perimeter would be enabling fuzzy skin.

Tinchus2009, 2024-09-17:
> This is with 2.5.59
>
> 10% Encroachment [...]

This is woorking as intended ans it is ok, but my particular example is different. Look at the posted picture, the encrochment setting seems to work only when you have internal infill and its connection to surrounding perimeters.  My case is a situation where I have an area with primeters and another area netxt to it and with a modifiers setting with 0 perimeters so that area has only infill. The conection between them is what you see on my picture and the encrotchment settings seems to no have effect at all

Tinchus2009, 2024-09-17:
> If I understand correctly, you have model consisting of two types of areas, one with perimeters and one without. Let's call the one with perimeters A and the one without B. You're trying to improve [...]

TThabk to you Y have learn how to use this kind of code on the slicer, didnt knew these kinds of settings were possible.
The idea looked good and I tested it. It doesnt work for 2 reasons: the 150% extrusion makes the perimeters look reaally ugly, and then it seems that the extra extrusion is also taken into account and so the connection of my picture is the same , because the infill is moved in order to not overlap with that perimeter.