supermerill/SuperSlicer #4415: Unexpected infill

Shine753, 2024-09-10:
**Bug description**
Trying to slice an stl, I see on layers 20 to 30 that there is a strange (and unexpected) infill that is generated on one part of the object. Fusing the moving part with the body. Can be seen in red in the bottom right of the joined picture.
A bug ? Or which parameter can cause that and how avoid it ?

**Expected behavior**
No infill. Sliced as the other similar parts on the object.

!!>> Project File <<!!
See zip file attached

**Screenshots**
See zip file attached

**Desktop (please complete the following information):**
Windows 10, Ender-3, SuperSlicer 2.5.59
[ss bug.zip](https://github.com/user-attachments/files/16947533/ss.bug.zip)

clarkschaefer, 2024-09-10:
I was able to fix this by adjusting the "Slice gap closing radius" setting under Print Settings > Slicing > Filtering to a value less than  half of the modeled gap; this is because the documentation notes that "cracks less than 2x the slice gap closing radius are being filled during triangle mesh slicing". The default setting is 0.049, which would automatically close any gaps less than 0.098 mm; this test model has a gap here of .05mm, which would be automatically closed. See slicing results below:

[image]

And the relevant setting:

[image]

I'm pretty sure this is a nominally a-okay setting for the overwhelming majority of models, but as people attempt to get finer and finer tolerances, it may be worthwhile reducing this default value, since small slices like this are showing up as printers improve.

neophyl, 2024-09-10:
There are several github issues on this on the Prusa Slicer github.  Its always someone wondering why their tolerance test parts are fused when slicing.  Its an outdated default setting (same in PS).  Its something I changed in my default profiles years ago.  I just use 0.005 for everything now.  If a model has that many issues that you need to use this setting to close gaps in it then it needs fixing probably anyway before printing.

Shine753, 2024-09-10:
Seems fine with this setting. Thank you.

Shine753, 2024-09-10:
Hmmmm.. Same model with the parameter set to 0.005, I can see on the borders that there are near-triangle shaped places, that seems the same :  

- some that are filled then non-filled on the next layer (see top-left or bottom right) 
- other ones that are filled and still filled on the next layer (see bottom-left)
- and other ones that are not filled at all... (see right and top-right)

[image]

Any idea from where this comes from and a way to avoid this ?

clarkschaefer, 2024-09-10:
[image]

That is almost certainly this setting for "Solid infill threshold area", automatically replacing very small isolated areas of infill with solid infill, since they are so small that conventional infill might not put any internal structure in there. Personally, I have it set to 0 because I find the setting not especially useful, and very small cross sections like that derive most of their strength from the perimeter anyway. Solid infill doesn't really add that much in my opinion.

Shine753, 2024-09-11:
Works fine, thank you.

closed 2024-09-11 by Shine753