supermerill/SuperSlicer #4416: New UI for Variable Layer Height

Shine753, 2024-09-11:
**Is your feature request related to a problem? Please describe.**
Variable Layer Height is a great feature. But it comes with a terrible UI.
Is this possible to replace this UI with another one. Or add a feature that makes it less headache to use.
I mean, when you want for example a default 0.2mm layer height, but exactly 0.1 from 0 to 5mm and exactly 0.3 from 48 to 49mm, it's a nightmare for such an easy wish.
If this can be done, I may take some time to propose some UI for this, and discuss around that first proposal.
Can this be done ?

**Describe the solution you'd like**
See above.

**Describe how it would work**
To be defined.

**Describe alternatives you've considered**
Not using Variable Layer Height

**Additional context**
To come...

neophyl, 2024-09-11:
Don't do that using VLH.  Do that using Height range modifiers.
Just add a series of them for each range and then the height to slice with is the basic setting on that modifier.
[image]

Shine753, 2024-09-11:
Great feature, didn't know that. Pretty similar to what I was starting to design :)
Just missing 2 things : the ability to tell that you want a smooth transition before / after the modifier, on xxx milimeters, using xxx mm steps. But as it is, it's already great :)

neophyl, 2024-09-11:
Yeah its handy.  If I was to do a redesign I would do it as part of a height range modifier.  So you define a height range and then one of the options on the modifier would be a discrete VLH for that range.  So you could set the VLH for 5-10mm say and then  add a different height range modifier and add a second VLH for that range. Etc.
Rather than it being 'global' for the whole model.

legend069, 2024-09-11:
> Don't do that using VLH.  Do that using Height range modifiers.
> Just add a series of them for each range and then the height to slice with is the basic setting on that modifier.
> [image]
> 

I didn't even know about this feature, thank you!