supermerill/SuperSlicer #4421: Add stop, change color, custom gcode also on the in layer slider

Tinchus2009, 2024-09-22:
**Is your feature request related to a problem? Please describe.**
Today we can create a change color stop, or add cutom gocode at a specifi layer change using the right slider after slicing. This is good for changing the color of a log for example, The bad: you change the color of the logo but also you change the color of the hole layer and this is noticeable on the perimeter for example.

But we have another slider that let us see how that layer progress. Im doing this now manually, but it would be more easy if we had the ability to add a custom stop, change color / add custom gcode, at a specific location using the bottom slider. In that weay we can change the color only starting at the infill for wexample, so a logo would have a different color but the main color would still be present on the outer object perimeter.

**Describe the solution you'd like**
Add the function os stopping/adding gcode the same as on the right slider to the bottom layer slider

[image]

legend069, 2024-10-09:
i'm wondering if the "region_gcode" feature would be better suited here?
using the modifier box you should be able to add the "tool change/color change" command per the region you want.

[image]
[image]

Tinchus2009, 2024-10-09:
I didnt knerw that option. I guess it can work. The problem I see is that would be super unconfortable and dificult since my proposal is add a change fillaent comand to just 1 layer. That modifier would need to be 1 layer thick and be positioned very very precise on the opbject nbd would not work in the case of a text fot example

legend069, 2024-10-09:
> I didnt knerw that option. I guess it can work. The problem I see is that would be super unconfortable and dificult since my proposal is add a change fillaent comand to just 1 layer. That modifier [...]

The ``region_gcode`` is relatively new.
It was added with my pressure advance calibration tool, but it should still work for this case.

You should be able to manually scale and type in the z position value for the modifier box.

I'm not 100% sure how it will process the filament change in a specific area, I want to hope that it'll process it the same, adding the command per object per layer.

Tinchus2009, 2024-10-15:
Hi. Ok, I tried to do what you proposed, an I think it doesnt work for different reasons: 1 the objective of this new option probably was other than this, but at least it is not adding the M600 correctly on the proper place, or it does it, it is just that user doesnt have exact control where. 2.- even if it did, it super dificult to do it, to palce the modifier in the correct place, at the correct Z height... super dificult.
Ill give an example off a print Im doing right now, it a pencil case for my sons, each of them has their names, and my intention is to stop the print in the are that fills the name´s in order that just the names has a different color and not the hole layer. Look at the pictures.  If that slider could give the user the ability to be moved and add a specif custom gcode at that location, the same as with the layer option slicer, the operation wpuld be super easy In the first image you can see the first stop and the firm M600 code, Im adding that code manually now editig the exported gcode and useing the interface to see the line number, search for it and add the M600 line

[image]

Then I move the slider till that section is finished, look at the line number, search for it adn add another extra M600 line. And then I repeat the process with the other objecto:

[image]

I think the feature could be very usefull and from the coding perspective since it is the same functionality as the layer slider has, dont think represents big troubles