supermerill/SuperSlicer #4428: Crashes Depending on Part Orientation

muckmed, 2024-09-27:
### Version
Occurs on 2.5.60.0 & 2.5.59.9, I assume other versions as well.

### Operating system type + version
Windows 10 Home Version 22H2 64 bit

### Behavior
Attempting to slice the attached file in certain orientations results in a crash. Occasionally, SuperSlicer will display the message shown in the photo below. 

[image]

### _**My temporary workarounds:**_
A. Orientate the part vertically before slicing. No longer crashes, but printing isn't feasible.

B. Import part -> Right click part -> Add part -> Box. Once the box is added the part will slice successfully in any orientation. HOWEVER, if the height of the added box is less than the height of the "Blower Fan Shroud" part, it will crash. The box must be at minimum 1 layer larger. In this case, on a .2mm layer height & default orientation, the Blower Fan Shroud part is 12mm tall after slicing. A box 12.2mm or taller will allow the part to be sliced. Box does not need to intersect the part. 

#### Project File (.3MF) where problem occurs
[Blower Fan Shroud.zip](https://github.com/user-attachments/files/17160612/Blower.Fan.Shroud.zip)

JohnEdwa, 2024-09-29:
I've found a few models as well that crash or fail to slice on 2.5.60.0 depending on orientation, maybe they can help nail down the issue. Trying to slice the calicat in the normal orientation throws that error and shortly afterwards force-closes the program, while trying to slice it standing on its head throws the same error, but without a crash, until you try again.

The last processing step before it freezes is always "Computing seam visibility areas: object 1/1.", and crash_log.txt is not generated.

[SS_2.5.60.0_orientation_crash.zip](https://github.com/user-attachments/files/17180075/SS_2.5.60.0_orientation_crash.zip)

starsolaris, 2024-10-01:
Another example 
[SH72_extension.zip](https://github.com/user-attachments/files/17213647/SH72_extension.zip)

supermerill, 2024-10-01:
I'll try to find a little time to work on the 2.5, to have a more stable stable release.

ExploiTR, 2024-10-04:
Same issue - crashing at compute seam visibility.