supermerill/SuperSlicer #4430: Exposed infill

jzagaja, 2024-09-27:
Old subject - using modifier or paint surface to expose some for infill.
Modifier adds perimeter unfortunately so only removes top surfaces.

[image]

legend069, 2024-09-27:
can you please explain this a little better?
i don't understand.

jzagaja, 2024-09-27:
Modifier crosses object and should just remove perimeter and leave anything else unaffected. It does but add perimeter elsewhere according to its shape. If you google "expose infill slicer" you will see lot of questions like this.

neophyl, 2024-09-27:
No it cant do that.  You still have the perimeter count from the object you are using the modifier on.  Its why when you add a modifier to add perimeters around a hole for example you get more than you think as you get the ones from the modifier AND the ones from the original object.

jzagaja, 2024-09-28:
So please add paint on surface to remove perimeter - quick and efficient. Many people requested that already.

legend069, 2024-10-01:
> quick and efficient.

it's not as easy as you would think..