supermerill/SuperSlicer #4432: Filament/Print Settings dropdowns scroll unnecessarily

PizzaScientist, 2024-09-29:
### What happened?

When I have a large number of filaments, print settings, (or even presumably printers), the dropdown organizes them all alphabetically, which makes sense. If I choose something at the top of the list (like Black), then chose it again, I can see all my filaments. But if I choose something at the bottom (like White), when I reopen the list, it shows the full size, but scrolled all the way to the bottom, with a lot of wasted space. I have to scroll back up, which makes it super annoying to switch back and forth between filaments (such as [B]lack and [W]hite)
-----------
[image]
-----------
[image]

### Project file & How to reproduce

1) have a large number of filaments
2) select a filament near the bottom of the list
3) try to select another filament
4) The dropdown choices will be scrolled off the top of the screen, even though they fit

### Version

2.5.59.8

### Operating system

Manjaro KDE kernel 6.6.47-1

legend069, 2024-10-09:
when selecting the last filament profile
would you prefer the window to remain the same size and just scrolled up to make the last filament profile be at the bottom of the list?

this might be a tricky feature to fix.

as an alternative you can re label your filaments with a number at the start(i'm pretty sure this will work) ie
 ```
1 Elegoo Yellow Rapid PETG
2 Elegoo Grey PLA+
3 Elegoo Blue PLA+
4 ect
```

PizzaScientist, 2024-10-14:
No, thats not quite the problem. The dropdown windows are always the correct size. The filament dropdown is always exactly big enough to fit your filaments (and the print settings dropdown is big enough to fit your print settings, and the printer dropdown, your printers) (within reason..... Not the case if I had a million filaments). The dropdown is the exact size needed, no bigger, no smaller.

The problem is that, if your last filament was at the bottom of the list, when you click the list again, it seems to try to show that filament at the middle of the dropdown window, like some kind of "active front and center" feature.

Imagine you had a binder, with pages in a page protector. And the protectors are the exact size of your paper. Now imagine that if you were reading a line of text at the bottom, and the page suddenly jumped half way out of the protector, so that the bottom line was in the middle of the plastic (active front and center), and the top half of the page was hanging out of the protector, and the bottom half of the protector was totally blank with no paper in it. And if you want a line of text at the top, you have to cram the page back in.

What a hassle! In the first screenshot, all the filaments would clearly fit into the dropdown in the grey area. if you measure it, the grey areas are the same size. 

I would like the filament dropdown to be as large as needed to show all my filaments (within reason), and for them to properly populate, only scrolling when necessary

I've included 2 more screenshots on the filament settings page. The issue happens there as well. Hopefully its clearer in these larger screenshots

Again, the issue isn't specific to filaments. It happens to all 3 dropdown (filaments, print settings, and printers). Its just more pronounced on the filaments, because I have way more filaments than I do print settings, or printers. 

[image]
[image]