supermerill/SuperSlicer #4436: SuperSlicer 2.5.60.0 crashes when slicing file

RobertHerhold, 2024-10-10:
### What happened?

On two separate files (one of which is probably a remix of the other), SuperSlicer crashes when I slice it. Here are two files that it repros with:

https://github.com/VoronDesign/VoronUsers/blob/main/printer_mods/sttts/Waveshare-5.5-inch-HDMI-AMOLED/STLs/Voron-2.4-Mount-Generisch-v6.stl
https://mods.vorondesign.com/details/mn2APihm7NkJZhoskam1CQ

### Project file & How to reproduce

[PITFT50_45_degree_mount.3mf.zip](https://github.com/user-attachments/files/17321017/PITFT50_45_degree_mount.3mf.zip)

### Version

2.5.60.0

### Operating system

MacOS Sonoma 14.6.1

### Printer model

Voron 2.4

Contrasens, 2024-10-14:
@supermerill @RobertHerhold I have probably the same issue: 2.6.60.0 crashes (no error, windows simply closes itself) at the end of the slicing process. It can be reproduced when slicing the **moon-globe-mk2.stl** from here: [https://www.printables.com/model/23859-designer-moon-lamp/files](https://www.printables.com/model/23859-designer-moon-lamp/files).

I tried the same in the latest PrusaSlicer 2.8.1 and it works just fine.

jodiecunningham, 2024-10-16:
Windows 11,  Kingroon KP3s, 
Happening to me on 2.5.60.0 as well. 

Supports or no supports, doesn't matter on this file
https://www.printables.com/model/330065-synthesizer-stand-45-degree-angle/

Seems like files I made in Tinkercad do just fine.  I can do a little more debugging later on my side.

RobertHerhold, 2024-10-21:
@supermerill thanks for taking a look at a fix. I'm super keen to try out the fix but MacOS is reporting "You can’t open the application “SuperSlicer” because it may be damaged or incomplete." from the [nightly build](https://github.com/supermerill/SuperSlicer/actions/runs/11347297937) and I'm having some issues building it from source. Do you know when it might make it into a release?