supermerill/SuperSlicer #4437: Object modifiers in 2.5.59.13 not as expected

CuredPrusa, 2024-10-13:
I doubt this is version related.
The object can be modified with infill (for example) just fine, as expected:
[image]
[image]

But when I go with fuzzy skin, it starts introducing perimeters of the modifier:
[image]
[image]
[image]

I would like to have just one side of the cylinder fuzzy. No other unwanted artefacts. I see that Orca is handling the same situation much better, but I can't use Orca because the Orca UI is driving me crazy with all those unnecessary options in settings, variations and subdivisions of subdivisions in profiles.  

Why is this happening with fuzzy skin in SS? Is there a workaround? Did I do something wrong?
Are there any other modifying options which would induce the same problem (so I can prepare myself)?

neophyl, 2024-10-13:
That’s exactly as expected with both PS and SuSi. Anything that involves different perimeters does that, as using a modifier splits it into separate slicing volumes so you get the perimeters from both. 

There are many issues on the PRUSA slicer GitHub about it and I’m sure if you searched here you will find more.

CuredPrusa, 2024-10-13:
Thanks for the answer. So I guess I should live with it.
How come Orca made it work so smoothly?
Orca slices my project in 7 seconds, SS in 15s (after I have deleted appdata folder when I was waiting for more than 40sec for 1,5hour print job to slice). Orca delivers 11MB file, SS 22MB. I guess that could be due to 76.912 instances of M106 commands in SS gcode, vs 294 instances of M106 commands in Orca.

I wish SuperSlicer maintains its strength in this powerful development of other slicers.  I can navigate and tweak SS settings blindfolded, I'd really like to stick to it. Orca's UI looks like deliberately messed up, for programmers to show off how smart they are. Software should be made for user, not just for a sole purpose of creation of it. I wonder if it's obvious I got recently frustrated by Orca. :)

CuredPrusa, 2024-10-15:
All clear. 
Horses for courses.

closed 2024-10-15 by CuredPrusa