supermerill/SuperSlicer #4439: Variable Layer Height Issue

giubu, 2024-10-17:
### What happened?

The maximum and minimum layer height sliders are stuck on 0.07mm
There is flickering on the slider and the value alternates very fast from 0 to 0.07
The actual value cannot be edited manually
Build Version: 2.5.59.6
System Version: Linux 

### Project file & How to reproduce

It happens with pretty much every file, sometimes it will adapt the layer height, but most of the times the layer height will be 0.07

### Version

2.5.59.6

### Operating system

Mx-Linux 23

### Printer model

Custom

stxve-cr3d, 2024-10-17:
It´s an old issue. Please update to the newer versions.

stxve-cr3d, 2024-10-17:
Also try to use % if it doesn´t work. 
Default:
- Min: 5%
- Max: 75%

stxve-cr3d, 2024-10-17:
It is a calculation error, try setting it in the settings to % as I said:
Min: 5%
Max: 75%

Printer Settings -> Extruder 1

giubu, 2024-10-18:
Thank you, this solved the issue.