supermerill/SuperSlicer #4443: Default Fan Speed is missing from the alternative GUI layout

NameOfTheDragon, 2024-10-19:
### Version
2.5.59.13

### Operating system type + version
Windos 11

### Behavior
When the GUI layout is set to the alternative style (not the default Prusa Slicer layout), the `Default Fan Speed` setting is missing from the `Filament` -> `Cooling` screen. I would expect this setting to appear as a Basic option in the arrowed location.

[image]

_Is this a new feature request?_
No - I think its a bug that basic settings are missing.

#### Project File (.3MF) where problem occurs
Not related to any specific project.

legend069, 2024-10-22:
iirc the default_fan_speed got moved try deleting the ``%appdata%\SuperSlicer\ui_layout`` folder and launch SuperSlicer again.

NameOfTheDragon, 2024-10-25:
> iirc the default_fan_speed got moved try deleting the `%appdata%\SuperSlicer\ui_layout` folder and launch SuperSlicer again.

That folder is empty on my system

NameOfTheDragon, 2024-10-25:
here is a PR that fixes the issue: #4464