supermerill/SuperSlicer #4448: Is there a way to define what should be considered as "top surface" ?

CuredPrusa, 2024-10-21:
In latest version (I guess the same goes for older) ironing is assigned to more than just top surface. Also for something that could look like a top surface. So I get outer perimeters ironed when in slope. And that is totally unacceptable. Is there a way I can tell slicer about min surface which should be considered as top surface?

[image]

CuredPrusa, 2024-10-22:
It certainly would help because it doesn't make sense to lose time and nozzle pressure just to run totally unwanted ironing on a perimeter line. On the other hand, if there would be long perimeter line, wouldn't it fit above minimum surface for ironing? I guess "surface width" would do better job than "area". Or both in tandem. Thanks!

legend069, 2024-10-22:
> On the other hand, if there would be long perimeter line, wouldn't it fit above minimum surface for ironing?

Ooh that is a good point!
Adding both options would be an improvement.

Thank you for the suggestion!

CuredPrusa, 2024-10-22:
But still, that would be just a workaround for something that Orca does just fine. Maybe you should take a peek at that code. I think it is related do the exposed perimeter area, same as the calculation of overhangs.

[image]

Nbeknel, 2024-10-22:
Have you looked into the ironing infill pattern for top solid layers?
[image]

Top benchy is with ironing PP, while bottom one is with ironing infill pattern for top solid layers, all other settings are the same. Their visualisation in the preview are considerably different even when using the same settings. However, notice the sloping areas.
[image]