supermerill/SuperSlicer #4449: Question: How to increase computing performance/speed of SuperSlicer?

CuredPrusa, 2024-10-21:
Hi
Can someone tell is there a trick to boost up the slicing performance of SS?
When I slice something simple, I don't notice anything, but as soon as I add a model with more complexity and a bunch of modifiers, SS slows down.  I would thought it's normal and I should live with it, until I tried Orca Slicer and got 8 seconds from the same parameters which SS computes 38 seconds.   If someone has an idea, I'm listening. Thanks. 

Version 2.5.60.0

legend069, 2024-10-22:
there are a fair few algorithms that have a heavy tax on CPU power.
can you please upload your config/project file.

mrx23dot, 2026-10-28:
No slicing issue on i5 gen2, please provide proof or close ticket