supermerill/SuperSlicer #4469: Superslicer crashes when I click on filament cooling tab

Odessa2433, 2024-10-29:
### What happened?

Sorry for any format errors, this is my first time reporting an issue.
Problem: Superslicer crashes when I click on filament cooling tab
What I Expected: To change the fan speed
Happens every time, With or without a model on the buildplate.
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [47597]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.5.59.0_2022-08-11 (SuperSlicer_2.5.59.0_2022-08-11)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-10-29 14:26:17.1740 -0400
OS Version:            macOS 14.2.1 (23C71)
Report Version:        12
Bridge OS Version:     8.2 (21P2057)
Anonymous UUID:        90121050-2301-F4AB-F7F8-FB85BFAE1B45

Time Awake Since Boot: 1300000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000128
Exception Codes:       0x0000000000000001, 0x0000000000000128

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [47597]

VM Region Info: 0x128 is not in any region.  Bytes before following region: 140737487797976
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      shared memory            7ffffff78000-7ffffff79000 [    4K] r-x/r-x SM=SHM  

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   SuperSlicer                   	       0x10a78e0ff Slic3r::GUI::OG_CustomCtrl::CtrlLine::update_visibility(Slic3r::ConfigOptionMode) + 47
1   SuperSlicer                   	       0x10a78e05d Slic3r::GUI::OG_CustomCtrl::update_visibility(Slic3r::ConfigOptionMode) + 93
2   SuperSlicer                   	       0x10a786ed2 Slic3r::GUI::ConfigOptionsGroup::update_visibility(Slic3r::ConfigOptionMode) + 82
3   SuperSlicer                   	       0x10a73fb99 Slic3r::GUI::Page::activate(Slic3r::ConfigOptionMode, std::__1::function<void ()>) + 393
4   SuperSlicer                   	       0x10a73662b Slic3r::GUI::Tab::activate_selected_page(std::__1::function<void ()>) + 123
5   SuperSlicer                   	       0x10a73ff98 Slic3r::GUI::Tab::tree_sel_change_delayed() + 808
6   SuperSlicer                   	       0x10a74b0df wxEventFunctorFunctor<wxEventTypeTag<wxTreeEvent>, Slic3r::GUI::Tab::create_preset_tab()::$_8>::operator()(wxEvtHandler*, wxEvent&) + 111
7   SuperSlicer                   	       0x10b73b37e wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
8   SuperSlicer                   	       0x10b73b0eb wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
9   SuperSlicer                   	       0x10b73b000 wxEvtHandler::ProcessEvent(wxEvent&) + 96
10  SuperSlicer                   	       0x10b5a81e2 wxScrollHelperEvtHandler::ProcessEvent(wxEvent&) + 34
11  SuperSlicer                   	       0x10b5b869f wxGenericTreeCtrl::DoSelectItem(wxTreeItemId const&, bool, bool) + 751
12  SuperSlicer                   	       0x10b5b3c2d wxGenericTreeCtrl::OnMouse(wxMouseEvent&) + 3117
13  SuperSlicer                   	       0x10b73a3cf wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 447
14  SuperSlicer                   	       0x10b73b10a wxEvtHandler::ProcessEventLocally(wxEvent&) + 90
15  SuperSlicer                   	       0x10b73b000 wxEvtHandler::ProcessEvent(wxEvent&) + 96
16  SuperSlicer                   	       0x10b5a81e2 wxScrollHelperEvtHandler::ProcessEvent(wxEvent&) + 34
17  SuperSlicer                   	       0x10b73b51c wxEvtHandler::SafelyProcessEvent(wxEvent&) + 12
18  SuperSlicer                   	       0x10b67c359 wxWidgetCocoaImpl::DoHandleMouseEvent(NSEvent*) + 89
19  SuperSlicer                   	       0x10b676e77 wxWidgetCocoaImpl::mouseEvent(NSEvent*, NSView*, void*) + 167
20  AppKit                        	    0x7ff80b39e891 -[NSWindow(NSEventRouting) _handleMouseDownEvent:isDelayedEvent:] + 4580
21  AppKit                        	    0x7ff80b317572 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 404
22  AppKit                        	    0x7ff80b3171c3 -[NSWindow(NSEventRouting) sendEvent:] + 345
23  SuperSlicer                   	       0x10b6653db -[wxNSWindow sendEvent:] + 123
24  AppKit                        	    0x7ff80baae212 -[NSApplication(NSEventRouting) sendEvent:] + 346
25  SuperSlicer                   	       0x10b5e5422 -[wxNSApplication sendEvent:] + 98
[... 319 lines cut ...]
USB Device: USB2.1 Hub
USB Device: STM32 Virtual ComPort
USB Device: USB2.0 Hub
USB Device: Mass Storage Device
USB Device: T2Bus
USB Device: Headset
USB Device: Apple T2 Controller
Thunderbolt Bus: Mac mini, Apple Inc., 47.5
Thunderbolt Bus: Mac mini, Apple Inc., 47.5

### Project file & How to reproduce

No specific file involved.

### Version

SuperSlicer_2.5.60.0_macos_240702.dmg

### Operating system

MacOS SONOMA 14.2.1 (23C71)

### Printer model

Monoprice MP10

Odessa2433, 2024-10-29:
Update
Minor sucess, same issue with different parameters.
I found the post regarding running a script to delete bed texture and size and was able to access the cooling tab. Now it crashes when I try to slice more than one stl file on the buildplate. 
I am trying to slice one door_frame.stl and three window_frame.stl it crashed immediately. It succesfully slices just the door frame, but not when I add one window frame. 
Actually seems to be  the window frame, as it will not slice one instance of it. I tried to upload the files, but it wouldn't let me select stl files so I added the Printables link 
(https://www.printables.com/model/1048105-pumpkin-house/files)
I was able to slice the door and three windows successfully before doing all of the steps above

Odessa2433, 2024-10-29:
Here is the crash log for just the window frame

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [48296]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            org.SuperSlicer/
Version:               SuperSlicer SuperSlicer_2.5.60.0_2024-07-01 (SuperSlicer_2.5.60.0_2024-07-01)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-10-29 15:36:13.7193 -0400
OS Version:            macOS 14.2.1 (23C71)
Report Version:        12
Bridge OS Version:     8.2 (21P2057)
Anonymous UUID:        90121050-2301-F4AB-F7F8-FB85BFAE1B45

Time Awake Since Boot: 1300000 seconds

System Integrity Protection: enabled

Crashed Thread:        13  slic3r_BgSlcPcs

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0xfffffffffffffffc
Exception Codes:       0x0000000000000001, 0xfffffffffffffffc

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [48296]

VM Region Info: 0xfffffffffffffffc is not in any region.  Bytes after previous region: 18446603336221749245  
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      shared memory            7ffffff78000-7ffffff79000 [    4K] r-x/r-x SM=SHM  
--->  
      UNUSED SPACE AT END

Thread 0:: slic3r_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff807a8fa2e mach_msg2_trap + 10
1   libsystem_kernel.dylib        	    0x7ff807a9de3a mach_msg2_internal + 84
2   libsystem_kernel.dylib        	    0x7ff807a96b62 mach_msg_overwrite + 653
3   libsystem_kernel.dylib        	    0x7ff807a8fd1f mach_msg + 19
4   SkyLight                      	    0x7ff80d1247b1 _CGSShapeSurfaceInline + 246
5   SkyLight                      	    0x7ff80d1d3898 SLSSetSurfaceBounds + 235
6   AppKit                        	    0x7ff80bfcf961 -[NSCGLSurface setSize:] + 63
7   AppKit                        	    0x7ff80b9b2339 NSOpenGLContextUpdateOffScreenViewSurface + 535
8   AppKit                        	    0x7ff80b4d6488 NSOpenGLContextAttachOffScreenViewSurface + 91
9   AppKit                        	    0x7ff80b9b19de __27-[NSOpenGLContext setView:]_block_invoke + 433
10  AppKit                        	    0x7ff80b201f28 NSPerformVisuallyAtomicChange + 134
11  AppKit                        	    0x7ff80b4d61f7 -[NSOpenGLContext setView:] + 139
12  SuperSlicer                   	       0x1096c7769 wxGLContext::SetCurrent(wxGLCanvas const&) const + 57
13  SuperSlicer                   	       0x1077c52f2 Slic3r::GUI::GLCanvas3D::load_preview(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const&, std::__1::vector<Slic3r::CustomGCode::Item, std::__1::allocator<Slic3r::CustomGCode::Item>> const&) + 82
14  SuperSlicer                   	       0x1078f6fb8 Slic3r::GUI::Preview::load_print_as_fff(bool) + 2952
15  SuperSlicer                   	       0x1078f7d26 Slic3r::GUI::Preview::reload_print(bool) + 134
16  SuperSlicer                   	       0x107965406 Slic3r::GUI::Plater::priv::set_current_panel(Slic3r::GUI::wxTitledPanel*) + 742
17  SuperSlicer                   	       0x107967714 Slic3r::GUI::Plater::priv::select_view_3D(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) + 276
18  SuperSlicer                   	       0x107940611 Slic3r::GUI::MainFrame::select_tab(Slic3r::GUI::MainFrame::ETabType, bool) + 433
19  SuperSlicer                   	       0x10796459e Slic3r::GUI::Plater::priv::on_slicing_completed(wxCommandEvent&) + 414
20  SuperSlicer                   	       0x109b332de wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
21  SuperSlicer                   	       0x109b3304b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
22  SuperSlicer                   	       0x109b32f09 wxEvtHandler::ProcessEvent(wxEvent&) + 41
23  SuperSlicer                   	       0x109b32db1 wxEvtHandler::ProcessPendingEvents() + 225
24  SuperSlicer                   	       0x109a8b8f7 wxAppConsoleBase::ProcessPendingEvents() + 119
25  SuperSlicer                   	       0x109b3cae0 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 48
26  CoreFoundation                	    0x7ff807baa576 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
27  CoreFoundation                	    0x7ff807baa49a __CFRunLoopDoObservers + 493
28  CoreFoundation                	    0x7ff807ba9a1c __CFRunLoopRun + 850
29  CoreFoundation                	    0x7ff807ba9102 CFRunLoopRunSpecific + 557
[... 395 lines cut ...]
}

Model: Macmini8,1, BootROM 2020.61.1.0.0 (iBridge: 21.16.2057.0.0,0), 6 processors, 6-Core Intel Core i5, 3 GHz, 8 GB, SMC 
Graphics: Intel UHD Graphics 630, Intel UHD Graphics 630, Built-In
Display: VA2359 Series, 1600 x 900, Main, MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 4 GB, DDR4, 2667 MHz, Micron, 4ATF51264HZ-2G6E3
Memory Module: BANK 2/ChannelB-DIMM0, 4 GB, DDR4, 2667 MHz, Micron, 4ATF51264HZ-2G6E3
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Jul 10 2023 12:30:19 version 9.30.503.0.32.5.92 FWID 01-88a8883
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB3.2 Hub
USB Device: USB3.2 Hub
USB Device: USB Receiver
USB Device: USB2.1 Hub
USB Device: USB2.1 Hub
USB Device: STM32 Virtual ComPort
USB Device: USB2.0 Hub
USB Device: Mass Storage Device
USB Device: T2Bus
USB Device: Headset
USB Device: Apple T2 Controller
Thunderbolt Bus: Mac mini, Apple Inc., 47.5
Thunderbolt Bus: Mac mini, Apple Inc., 47.5

CuredPrusa, 2024-10-29:
Try with this
https://github.com/supermerill/SuperSlicer/releases/download/2.5.59.13/SuperSlicer_2.5.59.13_win64_240701.zip
and see if the problem persists.

Odessa2433, 2024-10-29:
Will that work on a Mac with an Intel processor? I am still confused by the mac and mac_arm differences.

CuredPrusa, 2024-10-29:
Oh, sorry.
Try with this:
https://github.com/supermerill/SuperSlicer/releases/download/2.5.59.13/SuperSlicer_2.5.59.13_macos_240701.dmg

CuredPrusa, 2024-10-29:
Unfortunately I personally can't help you with mac issues.  Hopefully someone else will jump in to see your problem.

Odessa2433, 2024-10-29:
Thanks Cured. So far, so good. I've had a couple other crash issues I thought might have been due to messing around with thicker internal perimeters and scaling down models. I'll try to reproduce those too. But this seems to have worked.

On a side note, any chance you know why I can no longer use superslicer to preview sliced gcode files?
I used to be able to, but now gut a Prusa slicer error message.
Thanks again for your help.

Odessa2433, 2024-10-29:
Unfortunately, trying to use the newer versions is what got me here. I want all the new features, but navigating all the different editions (graph, bis, no name), I don't know what all that means. Then the nightly releases. It's just over myhead.
I prefer superslicer to all the others, and I'm no novice to printing, but the number of options, and lack of coding knowledge is where I get into trouble.
Since there's no auto-update, I click on the update link and often just pick the top one. I'd love to see a superslicer edition guide for dummies. 
I also wonder if having installed different editions(This happened when I chose graph) may have messed up the library.
Anyway, Thanks for you help. Working for now.