supermerill/SuperSlicer #4471: Mac OS Seqouia Update 15.1

Horusei, 2024-10-30:
### What happened?

After updating MAC OS, progranm cannot be started anymore.

The Program Finder does not have permission to open from "(null)"-< tranlated from german

### Project file & How to reproduce

Update Mac Os via System Settings

### Version

2.5.59.13

### Operating system

MacOs 15.1

Jarery, 2024-11-01:
I have the same issue. Ive gone through and deleted custom textures from other thread with no success.
I have tried changing permissions, tried opening from multiple apps such as finder, pathfinder, forklift.

No success.
Prusaslicer works fine

Creo2005, 2024-11-08:
> ### What happened?
> After updating MAC OS, progranm cannot be started anymore.
>
> The Program Finder does not have permission to open from "(null)"-< tranlated from german
>
> ### Project file & How to [...]

the same :(

Creo2005, 2024-11-08:
On macOS 15.1 need Signed and notarized snapshot build. 
so did openSCAD. And now openSCAD has started. See https://github.com/openscad/openscad/issues/880#issuecomment-2453281962

mszulen, 2024-11-10:
This should be your fix as mentioned by Jarery: terminal command : "xattr -cr /Applications/SuperSlicer.app"

Creo2005, 2024-11-10:
> This should be your fix as mentioned by Jarery: terminal command : "xattr -cr /Applications/SuperSlicer.app"

I know it and I did it. Doesn't help. MacOS Sequoia 15.1 Intel

Horusei, 2024-11-10:
> This should be your fix as mentioned by Jarery: terminal command : "xattr -cr /Applications/SuperSlicer.app"

I also tried and it didn‘t work. Changed Slicer.

Horusei, 2024-11-10:
Not resolved, Hit the Wrong Button. I already tried this an didnt‘t help. I changed the Slicer.

closed 2024-11-10 by Horusei

Vannue007, 2024-11-11:
Plus 1. “Xattr” don’t work and I’m using a M1 chipset