supermerill/SuperSlicer #4478: Default infill margin is broken in SS 2.7

vovodroid, 2024-11-05:
### What happened?

Option **Default infill margin** is broken in SS 2.7

[image]

[image]

It works fine in SS 2.5

### Project file & How to reproduce

[The Office Sign v2.zip](https://github.com/user-attachments/files/17628665/The.Office.Sign.v2.zip)

### Version

2.7.61

### Operating system

Win10

jbeima, 2024-11-16:
This has been moved to the Alpha Dev Team. We are not accepting any bug reports via GH at this time for 2.7.X,

Given this, this challenge was recreated and passed on.

CuredPrusa, 2024-12-16:
> We are not accepting any bug reports via GH at this time for 2.7.X,
> 
>

Is this related only to 2.7 or all versions in general?
Can someone from Dev Teams instruct users what to do with problems encountered during Super Slicer usage?
Should we bother to report issues or not? I see responses quite rarely.