supermerill/SuperSlicer #4480: max mm3/s being applied when it shouldnt

Tinchus2009, 2024-11-06:
### Version
_Version of SuperSlicer used goes here (help->about)_
2.5.60
### Operating system type + version
win11
### Behavior
I have set a print perofile. As you can see on screenshots, periimeter speed goes between 70 and 55 mm/s. Below you can see that there is a 10 mm3/s of extrusion volumen limit. That value is there because in the past that profiile used to use a value of 0 on perimeter speed in order to print using autospeed. But since I defined a value, autospeed should not apply anymore right? Seems it is, because while printing, I can see both external and internal perimeter being printed at a fixed 9.5 mm3/s of extrusion speed (around 45 mm/s of speed with a 0.6 nozzle)

[image]

[image]

[guia2ABS.zip](https://github.com/user-attachments/files/17647805/guia2ABS.zip)

neophyl, 2024-11-06:
Yes it should apply as that's the expected operation.  The volumetric limit is your safety break in effect.  You configure that at the maximum your extruder/filament combination can extrude at.  That way it can never go faster than your extruder can keep up with.

You don't want the safety then update your profiles to change it.  It is also why there are multiple places it can be set, it will use the lowest defined as that safety value.

Tinchus2009, 2024-11-06:
Ok, so the answer is "yes", the volumentrioc limit is being applied. In that case I think the help ballon should be changed because is says: 
[image]

In my case, the all values are not cero so auto speed should have not been applied following that description

neophyl, 2024-11-06:
I can't see the problem.  It says its a cap on the maximum flow rate, and as such is a cap on the Autospeed.  It itself is not a speed field, those are in the Speed for print moves section.  That section is the advanced and both the values in it are for setting some form of limit to the otherwise unlimited speeds/extrusions you can get when a speed field is set to zero and using autospeed.

It even says If this field is set to 0 then there is no autospeed or maximum flowrate except the printers maximum.  The meaning of the tooltip seems quite clear to me.

Tinchus2009, 2024-11-07:
The problem is that it is not working as intended on that help text: that flo limits should work on the speeds set with a cero value. If you see the screenshot of my config, the perimeters speed are set to 70 and 55. I never get taht speed, the speed on perimeters are being set by autospeed t 45 mm/s (and they are long perimeters, 200 mms long so there this is not a problems of short movements). So this meand despite I have set a defiined perimeter speed, the flow limit is being applied, but the help text says this limit only applies to the speeds set with a cero value

neophyl, 2024-11-07:
I disagree, that’s not how it reads to me. It’s a limit, it applies to everything. It says it’s a cap. First paragraph. You are treating the second paragraph as part of the first, it’s not. Can all the tooltips be improved, possibly. 
If you wanted to write an essay for each one. No matter how you word them there will always be someone that has trouble with them. Treat this as a learning experience and move on. Now you know that volumetric limits in both their locations act as a maximum safety limit and use your new gained knowledge to configure things properly.

mszulen, 2024-11-10:
btw, isn't the flow limited in the filament settings? There's also a volumetric speed limit field

Tinchus2009, 2024-11-11:
> I disagree, that’s not how it reads to me. It’s a limit, it applies to everything. It says it’s a cap. First paragraph. You are treating the second paragraph as part of the first, it’s not. Can all [...]

Well "move on" you say. I dont have a problem with that, despite you seem to just want to read first paragraph, so acording to you... lets just ignore second paragrap. Im not the developer of superscliver, so if I want to learn how to use it, I use the manual and if not in manual tooltips ar the other spurce. Tooltip here is very clear:  "autospeed is OPNLY enable if if speed fields has a value of CERO". It is simple readin for any kid. First paragraph is OK, secodn paragraph describes how it should work. Do you understand that or should I explain it slower?

It is a tooltip error? OK, i dont have a problem with that, I will just put on cero the cap and that will be all, I think there is no need for your crapy answer.