supermerill/SuperSlicer #4482: Crash when Generate Bed Level Calibration

michas79de, 2024-11-10:
### What happened?

Immediately after clicking `Calibration` - `Bed/Extruder leveling` and then `Generate` SuperSlicer crashes/closes without any warning.
Running from terminal, it says: "Speicherzugriffsfehler (Speicherabzug geschrieben)" which means in English "Memory access error (memory dump written)”, while crashing.
IDK where the dump is written or how to find it. So I couldn't attach it.

Slicing a STL or other calibration models seems to run fine.

Version:
I'm using the SuperSlicer-ubuntu_20.04-2.5.60.0.AppImage
Menu `Help` - `About` and the splash screen show version 2.5.60.0, but the window title and `Help` - `About` - `Copy` say 2.5.59.
Compiling my own app caused some errors in the end, but installed the same version 2.5.60.0 with the same crashing issue.

### Project file & How to reproduce

[Unbenannt.zip](https://github.com/user-attachments/files/17689918/Unbenannt.zip)

### Version

2.5.60.0

### Operating system

Ubuntu 24.04.1 LTS

### Printer model

Creality Ender 3 V2 with BL Touch added (changed G28 to G28G29 in start G-Code, but with only G28 it's no change in this crash behavior.)

mason-gordon, 2024-11-15:
I'm also getting the same crash with the stock Ender 3 V2 profile on Windows

legend069, 2024-11-23:
This is likely related to the German localisation (common issue) the "." And "," get swapped around.

As far as I'm aware the bed level calibration has not been changed in a while so this issue is likely still present in older versions.

michas79de, 2024-11-23:
I tried changing the language to English in SuperSlicer without success. The crash doesn't disappear.
I'll try changing the system language just for testing, but I won't change the whole system to English permanently.