supermerill/SuperSlicer #4486: Auto speed overrides modifier

CuredPrusa, 2024-11-19:
When I set certain speed at certain heigh range in layer modifier, it gets annulled by autospeed. I expect modifier should be above autospeed. Autospeed covers the whole model anyway, so without influence by modifier, the practical usability is very limited.
I've noticed only external perimeter speed obeys the modifier. 
Version 2.5.59.13

legend069, 2024-11-23:
Am I safe to assume you used the speed modifier for all extrusion roles?
Ie; internal infill, external/internal perimeter ect..

CuredPrusa, 2024-11-23:
I report issues as I find them when I work in SS.
This particular one I can not reproduce anymore and I have lost the track of the point where that happened.
I can not exclude my personal error.
So I'm closing the issue.
Thanks for support.

closed 2024-11-23 by CuredPrusa

legend069, 2024-11-23:
>This particular one I can not reproduce anymore and I have lost the track of the point where that happened.

This is the main reason we like project files :)