supermerill/SuperSlicer #4487: Issue with permissions on opening Superslicer on MacOs Sequoia.

Kwapin, 2024-11-21:
### What happened?

While opening the app it shows Spotlight has no permissions to open ("null"). And the app is not opening.
[image]

I have to add that it seems to be the problem created by apple and app signing. Search on the web shows that openScad had the same problems.
For the moment I have been opening the app by going straight into app packages/macOs.
And just now I have found solution that worked: 

By inputting into terminal:
codesign --remove-signature /Applications/SuperSlicer.app
codesign --force --deep --sign - /Applications/SuperSlicer.app

### Project file & How to reproduce

Just open the app

### Version

2.5.59.13

### Operating system

macOS 15.1

MilesTheGOAT, 2024-11-21:
I came here to post this exact issue, to add to this, I have built from source as well as downloading from the Releases.

When building from source I tried multiple different flags:
- Only build for Intel Mac
- Build for Mac with DMG creator enabled
- Built without additional flags and just running the script on it's own

When building from source I get a different error that the application file may be damaged or incomplete.

stxve-cr3d, 2024-11-21:
Please do xattr -cr Applications/SuperSlicer.app

Kwapin, 2024-11-21:
> Please do xattr -cr Applications/SuperSlicer.app Nothing more is needed.
It does not work for this. I tried. 

Also in my first post I provided the solution that worked for me.

stxve-cr3d, 2024-11-21:
xattr only works if the error is „app damaged“. In your case removing codesign was correct, merill got some problems with signing which will be fixes soon

sala91, 2025-04-28:
Any hope for fix?

sepo2025, 2025-04-30:
> ### What happened?
> 

> I have to add that it seems to be the problem created by apple and app signing. Search on the web shows that openScad had the same problems. For the moment I have been opening the app by going [...]
codesign --remove-signature /Applications/SuperSlicer.app 
codesign --force --deep --sign - /Applications/SuperSlicer.app

This worked for current MacOS and SuperSlicer version. Thanks!