supermerill/SuperSlicer #4500: Crashes upon slicing model

rsforester, 2024-12-03:
### Version 2.5.60
_Version of SuperSlicer used goes here (help->about)_

### Operating system type + version
_What OS are you using, and state any version #s_
windows 11 home

### Behavior
* _Describe the problem_
* crashes when slicing file when gets to seam visability
* _Steps needed to reproduce the problem_
* slice file
* _Expected Results_
* _Actual Results_
* crashes and closes
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_
[dock for ktouh dock.zip](https://github.com/user-attachments/files/17999344/dock.for.ktouh.dock.zip)

Dids, 2024-12-05:
I'm also getting this on seemingly random models (STL) and projects (3MF), however I'm on macOS. Also on version 2.5.60, but will try downgrading to see if that helps for now. That said, this crashing only just happened, and it's been working fine for a long time, so not sure why this would suddenly start happening.

rsforester, 2024-12-05:
I just did as you did and tried on 2.5.59.13 and did not crash. I am in the same boat as it worked for a long time and all the sudden had problems

jbeima, 2024-12-07:
Are your seams painted on? If so this is a known bug it 2.5.60.0...