supermerill/SuperSlicer #4504: exporting/importing problem

Chazy12, 2024-12-15:
### What happened?

when i try to export the configuration bundle.ini from superslicer and import it in prusaslicer gave me error.
[image]

### Project file & How to reproduce

file - export - bundle with physical printer

### Version

2.5.60

### Operating system

windows 11

jbeima, 2024-12-17:
This has been for a long time. This is more of a Prusa Slicer issue than us.

Respectfully of Prusa, this probably this should be submitted to the Prusa Team to correct in their import code. :)

Chazy12, 2025-01-03:
did it by myself.
i don't know if could help to anyone..

https://github.com/user-attachments/assets/4586e856-3d63-43ff-bfa8-783a9fbe71bb