supermerill/SuperSlicer #4513: Crash on model slice (segmentation fault (core dumped)

rab1515, 2025-01-04:
### Version
`Version: 2.5.60.0`
`Build: SuperSlicer_2.5.60.0_2024-07-01`

### Operating system type + version
Operating System: Unix
System Architecture: 64 bit
System Version: Linux 6.12.4-1-MANJARO x86_64

### Behavior* 
Superslicer crashes during model slicing. Using version `SuperSlicer_2.5.59.13_2024-07-01` does not crash but reports the following error:
[image]

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
[Superslicer2.5.59.13_linux_64_error_project.zip](https://github.com/user-attachments/files/18308592/Superslicer2.5.59.13_linux_64_error_project.zip)

rab1515, 2025-01-05:
Setting `Thin walls = false`  stops SuperSlicer from crashing. The issue is prevalent in the app image build and window build. The 2.5.59 builds are more graceful and display an error, while the 2.5.60 builds crash.

agittins, 2025-01-21:
I also get regular crashes using 2.5.60 on debian sid. Clicking generate in the bed levelling calibration dialog is enough to reproduce, it appears to happen at the slicing stage.

This still happens on my system if I disable Arachne and Thin Walls. Running a backtrace in gdb gives me:

```gdb
Thread 25 "slic3r_tbb_5" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffa3bfe6c0 (LWP 1031149)]
0x00005555563b1c9f in Slic3r::SeamPlacerImpl::calculate_polyline_angles_at_vertices(Slic3r::SeamPlacerImpl::PolylineWithEnd const&, std::vector<float, std::allocator<float> > const&, float)
    ()
(gdb) bt
#0  0x00005555563b1c9f in Slic3r::SeamPlacerImpl::calculate_polyline_angles_at_vertices(Slic3r::SeamPlacerImpl::PolylineWithEnd const&, std::vector<float, std::allocator<float> > const&, float) ()
#1  0x00005555563b5983 in Slic3r::SeamPlacerImpl::process_perimeter_polylines(Slic3r::SeamPlacerImpl::PolylineWithEnd const&, float, Slic3r::LayerRegion const*, Slic3r::SeamPlacerImpl::GlobalModelInfo const&, Slic3r::PrintObjectSeamData::LayerSeams&) ()
#2  0x00005555563bba85 in Slic3r::SeamPlacer::gather_seam_candidates(Slic3r::PrintObject const*, Slic3r::SeamPlacerImpl::GlobalModelInfo const&, Slic3r::SeamPosition)::{lambda(tbb::detail::d1::blocked_range<unsigned long>)#1}::operator()(tbb::detail::d1::blocked_range<unsigned long>) const [clone .isra.0] ()
#3  0x00005555563bc244 in tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<unsigned long>, Slic3r::SeamPlacer::gather_seam_candidates(Slic3r::PrintObject const*, Slic3r::SeamPlacerImpl::GlobalModelInfo const&, Slic3r::SeamPosition)::{lambda(tbb::detail::d1::blocked_range<unsigned long>)#1}, tbb::detail::d1::auto_partitioner const>::execute(tbb::detail::d1::execution_data&) ()
#4  0x0000555556fae00b in tbb::detail::r1::market::process(rml::job&) ()
#5  0x0000555556f9e3d7 in tbb::detail::r1::rml::private_worker::thread_routine(void*) ()
#6  0x00007ffff6dbd732 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:447
#7  0x00007ffff6e382b8 in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78

```

mszulen, 2025-01-21:
2.5.60 crashing for me on Mac. Might be similar reason. Thin walls doesn't make any difference.
It starts to slice and crashes around, what looks like, beginning of infill calculation.

[MacBook M1 SS Crash.3mf.zip](https://github.com/user-attachments/files/18495029/MacBook.M1.SS.Crash.3mf.zip)

0xUnkn0wN, 2025-02-12:
Same for me with 2.5.60 on Fedora Linux, if you use Arachne instead of Classic as perimeter generator no crash happens.

I attached one STL that always fails with Classic but not with Arachne.

[Crash_Part.zip](https://github.com/user-attachments/files/18764081/Crash_Part.zip)