supermerill/SuperSlicer #4521: Arc 2.5.59.13 vs 2.5.60.1

CuredPrusa, 2025-01-27:
I have noticed 2.5.59.13 generates much more G2/G3 than latest 2.5.60.1. 
Same resolution and other related settings.
How come?

supermerill, 2025-02-03:
I don't remember what I've done in 2.5.60.
I did at some point an algorithm to try to average the center of arcs over multiple points (if it keeps the arcs within tolerance), to be able to merge arcs that should be the same. I think I did it in 2.7, but maybe it was in 2.5.

CuredPrusa, 2025-02-03:
Nice to see you alive!
Well, I don't know either what should be the difference, but the truth is when I want to use arc benefit on my Marlin, I use 2.5.59.13 because G2/G3 in 2.5.60.1 (nightly from December 2024) is not sufficient.  It's not problem for me to switch, but I wish I could use the faster processing power of latest iteration of slicer. Any recommendations? Or should I continue as usual?

CuredPrusa, 2025-02-03:
> Ho, you mean that the arcs are less detected?

Exactly!
As I said, the same settings are in both versions. But even when I change resolution I don't see much increase in arcs in  2.5.60.1.

Ok. I'm waiting for 2.7 then. Where should I find download when it will be ready for testing?

jbeima, 2025-02-09:
We will post it here and on the Website.. The links on the website just point back here anyways.

The website will also have more on it for the release in the form of blog posts.

supermerill, 2025-03-26:
reopen if arcs still have issues to be constructed.

closed 2025-03-26 by supermerill