supermerill/SuperSlicer #4523: Image missing from polyhole page

RickSeiden, 2025-02-05:
The [polyhole](https://github.com/supermerill/SuperSlicer/wiki/Polyholes) page is showing an image not found error.  I got the gist of the page and I think the image was supposed to be the drill bit picture.

clarkschaefer, 2025-02-05:
[image]

Per search results for "polyhole", it does appear to be the drill bit picture. I am unfamiliar with how storing images works in github wikis, but I'd be happy to create a replacement image for use in SuSi's wiki.

supermerill, 2025-03-26:
Thanks for the report.
Seems like imgur decided to delete my picture.
made another one real quick .
If you want to create some picture of irl use cases, i can add one or two of them.