supermerill/SuperSlicer #4524: Crossing perimeter not triggered when moving to another model instance

CuredPrusa, 2025-02-06:
### What happened?

I don't know if this is version related. 
This is first time I tried printing with multiple instances at once /in the same plane. 
I have noticed totally unexpected behaviour: Even though moving to another object is actual crossing perimeter, SS does not retract nor lift when moving to other object. If I disable "only retract when crossing perimeters" it works fine, but this is not what I want.

Did I miss some additional setting? 
Is it version related?

### Project file & How to reproduce

//

### Version

2.5.60.1

### Operating system

W10

### Printer model

//

CuredPrusa, 2025-02-07:
[Shape-Box.zip](https://github.com/user-attachments/files/18704894/Shape-Box.zip)
Here is the test project.
I have realized there is a bug related to support structure. 
Whenever the travel is related to support, there is a chance it will not retract/lift. 
I would appreciate info about the version where this is corrected, so I can continue my project there. Thanks.

clarkschaefer, 2025-02-08:
[image]

Hmmm, I'm not sure which spot you're referring to. Looking at the gcode that 3mf produces, I see retractions (and surprisingly long wipes, I only do maybe 1/5 that distance) leaving and going to supports from normal perimeters everywhere I checked. I'm game for as many pictures as you can send of the issue, and would gently suggest reducing your wipe distance as well to see how that changes the print.

[image]

~One other setting in this bundle that I found was causing strange issues on other printers I run was "decelerate with target acceleration". As near as I can tell, this appears to be setting the new deceleration too late for the printer to decelerate from its current speed down to the target speed and causing interesting motion in the meantime. If you trust your travel acceleration capabilities, it is probably safe to disable this setting.~

Disregard, the distance does actually seem to be just enough... maybe the issues with my printer were all my own. 😅

If that doesn't solve it, I implore you to keep sharing images and knowledge; I would really like to get to the bottom of this.

CuredPrusa, 2025-02-08:
Wipe is here for disobedient fast running PETG.  It doesn't influence retractions in this case.
Lift is purposely exaggerated for better observation. 

Observe the travel lift here. 
The lines which are not lifted are not retracted as well. 

[image]

Once the support disappears, all travels crossing perimeters are lifted/retracted.

[image]

clarkschaefer, 2025-02-09:
Ope, that was my problem, I forgot to make sure I was loading this in 2.5.60. Confirmed it's not happening in 2.5.59.13 at least. I haven't touched SuSi in so long that I don't have the latest alpha (downloadable for your respective platform from here: https://github.com/supermerill/SuperSlicer/actions), but I also managed to confirm that it does not have these retractionless/liftless travels in 2.5.61.0/2.7 alpha, see below:

[image]

Chatter on the Discord suggest a normal public build will be available in the coming few weeks.

CuredPrusa, 2025-02-09:
Nice, thanks for heads up, appreciated.

CuredPrusa, 2025-02-09:
Is this "wipe in the air" attempt? Looks strange...

[image]

CuredPrusa, 2025-02-09:
I would stay with 2.5.59.13 until newer version becomes bullet proof, but slicing there takes way too long for my project.

jbeima, 2025-02-09:
The public Beta, is planned for Feb 14th... However that might slip due to there being no working version for the Mac and a focus on seeing if we can properly sign a Windows version to not trigger the even more aggressive Win11 defender.

jbeima, 2025-02-09:
We have put a lot of work into travels with 2.7.61.0 Alpha. If you think you had a lot above, you should of seen some of the earlier Alphas. lol