supermerill/SuperSlicer #4525: Little steps on flat walls

MLiv79, 2025-02-09:
### What happened?

This is a small box with totally flat walls, as you can see from the pictures attached i've on the preview and then on the print small steps on those who should be totally flat walls. With aracne perimeters generator less, with classic more. What could be the cause of this behaviour on external perimeters? Some mistaken settings? Speed, width and height are stable on the whole wall.

[image]

[image]

### Project file & How to reproduce

Slice something

### Version

2.7.61 alpha

jbeima, 2025-02-09:
We are aware of this issue already. :)

jbeima, 2025-02-09:
If you go into your printer settings and then disable both "Max G1 Per Second" and "Min Extrusion Length" this should go away.

However the Alpha is NOT supported. We do have the the next planned public Beta coming shortly.