supermerill/SuperSlicer #4532: Some clarification needet ,some requests and one issue.

printicus, 2025-02-22:
First... I searched an did not find what I need to know about the "extrusion spacing". I understand what it does, but what is the "normal" value that any other slicer is using? Second: I cannot find the spacing setting for infil and first layer. Why do I need spacing set to 89.275% for the infill that is one line and alotta space and then another line? Does that not mess whit the calculation of infill procentage? And the spacing for the first layer (wich is allso set to 89.275% for some reason) is messing whit the first layer quality. With the same slicer settings set in Prusa Slicer i got perfect first layers and for Super Slicer i need to rise the z offset by 0.06 to 0.08 mm to get the same results.
And one requests, if its possible. In the Filament Settings tab, under Cooling is a setting called "Dissable fan for the first". I wonder if this can be implemented for the bed and nozzle temperature, like the tranzition from the first layer temperature to the other layers themperature to be acros 2-3 or whatever i want numbers of layers.
My Issue: I have Kamp installed for adapting meshing (tryed the klipper one, for some reason does not work for me) and when i tryed to print with "Complete individual objects" setting enabled it does the messhing only for the first object that it prints.
Thank You.
My hardware: Voron 2.4 with Manta M8P and CB1.

jbeima, 2025-02-25:
This should be asked on the Slicer's Discord.

Not a bug...

printicus, 2025-02-28:
> This should be asked on the Slicer's Discord.

Isnt this the place for that? You bottered to answer, could you not give the whole answer?