supermerill/SuperSlicer #4538: Crash sometimes (seam involved?)

Henyckma-Sensei, 2025-02-27:
### What happened?

When I try to laminate a file sometimes, if it is large and contains difficult geometry, it crashes

I have noticed that it usually happens when I use the seam paint tool or when it is in the loading process it always stops computing visibility of the seams

### Project file & How to reproduce

1 Open the archive
2 generate gcode
3 the App crash

[crash test.zip](https://github.com/user-attachments/files/18999804/crash.test.zip)

### Version

2.5.60

### Operating system

windows 10

### Printer model

Ender 3

Henyckma-Sensei, 2025-02-27:
I just confirmed that in the previous version (2.5.59.13) it does not happen

imonsei, 2025-03-20:
I have a rather simple model that also makes it crash. I don't use paint seams, but "Nearest".
Also on v 2.5.60.00
Also crashes on "calculating seam visibility areas"

[glass beads ball bearing-cage.001.stl.zip](https://github.com/user-attachments/files/19360833/glass.beads.ball.bearing-cage.001.stl.zip)