supermerill/SuperSlicer #4542: cylinder holes missing

username-sucks, 2025-03-01:
superslicer started good until you hace a cyl or through holevertical turn print horz ok but cant print like that
whats going on ?

1   ORIGNAL
2   INFILL 75 % SOLID LAYERS  TOP & BOTTOM = 0
3  INFILL 75  %  SOLID LAYERS  TOP & BOTTOM = 0

prints ok in PrusaSlicer INFILL 100 %   SOLID LAYERS  TOP & BOTTOM = 4

and fill gaps & holes is not on
what happen to rename pre-set

[image]
[image]
[image]
[image]

 button ?

neophyl, 2025-03-03:
What version are you using ?  Check the slicing mode is still on Regular.  There's a mode on Print Settings>Slicing>Slicing mode called 'close holes' which does exactly what you picture.

If you had included basic info such as a saved SuSI project along with the version we could check the settings rather than just guessing.

username-sucks, 2025-03-04:
hi thanks for getting back to me
Regular. i did say that 'close holes' not on first think i checked i used
casic
now to make it all simple here is a few cyl different size
so i can tell you
smaller holes are problems when holes are 20mm or less bigger than 20mm
looks ok at
this is how do it with everything
infill 75% solid layers both top & bottom 0 bigger than 20mm ok     /
smaller holes no
infill 75% solid layers both top & bottom 3 bigger than 20mm ok     /
smaller holes no
Tom

neophyl, 2025-03-04:
Again as requested please attach a saved SuSi project file.  With that we can all see the settings you are using and can look for anything that might be causing it.  You also haven't even said what version you are using still.  
Slice to get the problem, then File>Save Project As.  Take the resulting .3mf file and ZIP it up.  Then attach it to a reply here.
Thats the minimum that the Dev's or anyone else needs to even begin looking at the problem.

There have been problems in the past with corrupted profiles, where settings from one type get accidentally saved into a different profile type, say filament.  When that happens you can set the value in the normal place all you like because there's a second one stored that you cant change.  
Not saying that's the case this time but the only way to tell is with a saved project.

username-sucks, 2025-03-04:
hi
i
did send cyl 1.3mf and cyl 2.3mf files or i dont what you mean
files are pretty well basic just renamed to what ever
it does for all filements
i do it again
Tom

neophyl, 2025-03-04:
I cant see **ANY** attached files to the report.  Please make sure you zip them up or they simply wont be attached.

username-sucks, 2025-03-04:
hi
ok sorry
Tom