supermerill/SuperSlicer #4543: This STL crashes SuperSlicer every time

sixtyfive, 2025-03-05:
### What happened?

Working on an enclosure, which is a little more complex. Once I hit "Slice" and "Generating G-Code" appears, a second later SuperSlicer crashes.

[amp case.zip](https://github.com/user-attachments/files/19079353/amp.case.zip)

### Project file & How to reproduce

I've tried playing with various settings (disabling object labels, disabling support, etc.), no luck so far. One thing I noticed is that it takes a little longer to crash with support enabled and skirt/brim disabled.

### Version

2.5.60

### Operating system

Windows 10

### Printer model

Ender-3 V2 with a custom dual z-axis mod and running Klipper

darco415, 2025-03-06:
I guess I was bored, d/l your .stl and ran on latest dev 2.7.6.1 and enabled organic support. Did not crash

sixtyfive, 2025-03-06:
Thank you for checking it! May I ask where you downloaded the dev build from?

clarkschaefer, 2025-03-06:
https://github.com/supermerill/SuperSlicer/actions this page has the latest nightly builds, ctrl+F for your OS platform, if you're on Windows don't forget to grab the deps (dependencies) as well. Good luck!

closed 2025-03-06 by sixtyfive