supermerill/SuperSlicer #4545: Add option "avoid crossing Ironed surfaces"

Arthur-de-Partuur, 2025-03-05:
**Is your feature request related to a problem? Please describe.**
Currently there is an option to "avoid crossing perimeters". This does not take (finished) surfaces into account. I see nozzle-skid-marks on Ironed surfaces that might have been avoided/

**Describe the solution you'd like**
At the Ironing interface, add a tick mark with "Avoid traveling over ironed surfaces"
This would re-route the nozzle away from a finished surface or lift the nozzle up a bit to avoid scratching.

**Describe alternatives you've considered**
Maybe a more general approach could be to add a "Avoid finished or finalised surfaces"

supermerill, 2025-03-25:
Do you have an image of the wrong travel that goes over the ironing?
I can think of multiple issues, and i'm not sure the one you encounter.

Arthur-de-Partuur, 2025-03-27:
Mmm, do you mean Images from the slicer, or an actual photo of a print?

supermerill, 2025-03-27:
From the slicer, showing the bad travels and with the project file so I can try myself.

Arthur-de-Partuur, 2025-03-28:
I had to look reallly hard to see if I can show you. I have made a screen recording. I start the recording at the end of the ironing and togle with my cursor a bit so you can see. It travels over the ironed surfaces and leaves a mark.

https://github.com/user-attachments/assets/9186124d-bc80-4e73-9691-ebcc9b683bfc