supermerill/SuperSlicer #4546: Super Slicer Crashes on Randomly and will not Re-open without reboot

Tegguy, 2025-03-07:
### What happened?

When importing a model superslicer will close without any error messages and will not re-open without a windows reboot. When I try to re-open the application starts but before the 3D viewer loads it closes again.

### Project file & How to reproduce

Importing file... has happened with multiple files and is not consistent. 

### Version

2.5.59.13

### Operating system

Windows 11

### Printer model

Voron 2.4

supermerill, 2025-03-25:
if you rename your configuration directory (C:\Users\my_name\AppData\Roaming\SuperSlicer), does it solve the issue?
if yes, can you give me a zip of your configuration?

Tegguy, 2025-03-28:
This did not fix it. Superslicer opened and when I tried to import a model
I had imported before it crashed again. When trying to re-open it loads but
before it shows the print bed it closes

supermerill, 2025-03-28:
you can give me the problematic project, to double-check.

Tegguy, 2025-03-28:
So I think I narrowed it down a bit more and it might not be a superslicer
issue but I'm not sure... it seems to happen most of the time when I plug
in my USB-C docking station but if I boot the computer with it hooked up
there is no issue.

Tegguy, 2025-03-30:
I stand corrected super slicer just closed on me without changing the
docking station status

supermerill, 2025-04-01:
Do you have ecc memory in your computer?

supermerill, 2025-04-01:
no error correction on laptop.
My guess is your computer may have a memory chip gone bad and causing random errors.

Tegguy, 2025-04-01:
Thank you for the support

closed 2025-04-02 by supermerill

supermerill, 2025-04-02:
@Nbeknel if you still have the buggy configuration, please give it to me to test it, to try to reproduce the error.

Nbeknel, 2025-04-02:
I've had it happen when I had both SuSl 2.5 (or 2.6) and SuSl 2.7 saving profiles to the same configuration folder. I've since added separate configuration folders next to the .exe for nightlies and beta/rc versions (as mentioned [here](https://github.com/supermerill/SuperSlicer/discussions/1772#discussioncomment-1577050)). I don't think I have kept the buggy configuration file.