supermerill/SuperSlicer #4548: 2.7.61.0 alpha - strange top fill resolution

CuredPrusa, 2025-03-22:
I have noticed when concentric TOP FILL is created, at certain diameter the resolution gets cut in half.  I wonder if that is normal.

[image]

Are there some additional settings? I don't see problems on finished print, but it looks shuddering  on screen and I see that display corresponds to the gcode. Version 2.5 renders the same model with fine top fill lines, same settings. Although, opening this 2.7 project in 2.5 makes 2.5 crash.  

[image]

SuperSlicer
Version:   2.7.61.0
Build:     SuperSlicer_2.7.61.0-2025-03-18

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 34,049MB

[test1.zip](https://github.com/user-attachments/files/19401286/test1.zip)

macchinette, 2025-03-22:
You could try this.

[image]

[image]

CuredPrusa, 2025-03-22:
Oh, right! 
I was mislead with 0.1 resolution in 2.5, it gives fine results there as seen on the screenshot. I guess 2.7 is more sensitive in that matter. 

Thanks for this.

closed 2025-03-22 by CuredPrusa

supermerill, 2025-04-02:
mmm.
I think there's a need to review it.
Maybe use the main resolution for top infill, and decrease the default value a bit.