supermerill/SuperSlicer #4550: macOS build for 2.7 completely broken

julian-weinert, 2025-03-25:
### What happened?

The build process seems to be completely broken.
Downloaded assets contain a broken application bundle with files missing and dynamic linking apparently having failed.

```
$ /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
dyld: Library not loaded: @executable_path/../Frameworks/
  Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
  Reason: no suitable image found.  Did find:
	/Applications/SuperSlicer.app/Contents/MacOS/../Frameworks/: not a file
	/Applications/SuperSlicer.app/Contents/MacOS/../Frameworks/: stat() failed with errno=1
	/Applications/SuperSlicer.app/Contents/MacOS/../Frameworks/: not a file
	/Applications/SuperSlicer.app/Contents/MacOS/../Frameworks/: stat() failed with errno=1
	/usr/local/lib/: not a file
	/usr/local/lib: not a file
	/usr/lib/: not a file
	/usr/lib: not a file
Abort trap: 6
```

### Project file & How to reproduce

1. Download 'SuperSlicer_2.7.61.0_macos_intel_250325.dmg'
2. Drag app bundle to /Applications/
3.1 Right click → Open
4.1 Confirm unknown developer
5.1 SuperSlicer crashes immediately

Run SuperSlicer from the Terminal to get more details
```
$ /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
```

### Operating system

macOS 10.14.6 (18G9323)

stxve-cr3d, 2025-03-25:
Can u try this commands:

brew reinstall icu4c

brew upgrade node

Also I‘m not sure if it‘s the same on intel version, but you would have to do a ‚sudo xattr -cr SuperSlicer.app‘ in the folder you have dragged the app (usually Applications)

supermerill, 2025-03-25:
I wonder about something:
when your error message say 'no suitable image found'
Is it about libzstd1.5.7.dylib?
do you have `libzstd` womewhere in `/usr/lib/` ?
For arm, I packaged it inside the dmg as it's not in the os, but it is in the intel version... unless it was removed in recent macos release.

julian-weinert, 2025-03-27:
I'm running an ancient version on an intel processor. 10.14 Mojave.
There is no `libzstd` in that location. But in `/usr/local/lib` I did find it.

supermerill, 2025-03-27:
can you `otool -L /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer` and paste the result here ?
(the dmg need to be loaded so the path to /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer exists)

ChasonDeshotel, 2025-03-27:
I have the same issue. Here's the full stacktrace

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [2605]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            org.SuperSlicer/
Version:               SuperSlicer SuperSlicer_2.7.61.0 (SuperSlicer_2.7.61.0)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-03-27 11:33:34.8027 -0500
OS Version:            macOS 13.6.8 (22G820)
Report Version:        12
Anonymous UUID:        5926319D-0E4D-047F-E56A-F3D942CF92A4

Time Awake Since Boot: 5300 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @executable_path/../Frameworks
Referenced from: <6B837830-B4D3-3542-A28F-39527F22E5C9> /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/Applications/SuperSlicer.app/Contents/Frameworks/' (not a file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	    0x7ff80eb44ae6 __abort_with_payload + 10
1   dyld                          	    0x7ff80eb5eda7 abort_with_payload_wrapper_internal + 82
2   dyld                          	    0x7ff80eb5edd9 abort_with_payload + 9
3   dyld                          	    0x7ff80eae3840 dyld4::halt(char const*) + 375
4   dyld                          	    0x7ff80eae0b66 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 4526
5   dyld                          	    0x7ff80eadf3b3 start + 1795

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7bee784e8  rdx: 0x00007ff7bee78950
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7bee78530  rsp: 0x00007ff7bee784e8
   r8: 0x00007ff7bee78550   r9: 0x0000000000000000  r10: 0x000000000000006d  r11: 0x0000000000000246
  r12: 0x000000000000006d  r13: 0x00007ff7bee78950  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x00007ff80eb44ae6  rfl: 0x0000000000000246  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x101086000 -        0x104fb1fff org.SuperSlicer/ (SuperSlicer SuperSlicer_2.7.61.0) <6b837830-b4d3-3542-a28f-39527f22e5c9> /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
    0x7ff80ead9000 -     0x7ff80eb713af dyld (*) <2b649d59-89d8-3db6-9ba4-a6aecba42f6e> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 0
    thread_create: 0
[... 141 lines cut ...]
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1600.157.0)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libbz2.1.0.dylib (compatibility version 1.0.0, current version 1.0.8)
	/usr/lib/liblzma.5.dylib (compatibility version 6.0.0, current version 6.3.0)
	@executable_path/../Frameworks/ (compatibility version 1.0.0, current version 1.5.7)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1336.61.1)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.12)
	/usr/lib/libcurl.4.dylib (compatibility version 7.0.0, current version 9.0.0)
	/System/Library/Frameworks/AudioToolbox.framework/Versions/A/AudioToolbox (compatibility version 1.0.0, current version 1000.0.0)
	/System/Library/Frameworks/WebKit.framework/Versions/A/WebKit (compatibility version 1.0.0, current version 617.1.17)
	/System/Library/Frameworks/Security.framework/Versions/A/Security (compatibility version 1.0.0, current version 61040.61.1)
	/System/Library/Frameworks/Carbon.framework/Versions/A/Carbon (compatibility version 2.0.0, current version 170.0.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 24.0.0)
	/System/Library/Frameworks/QuartzCore.framework/Versions/A/QuartzCore (compatibility version 1.2.0, current version 1.11.0)
	/System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/CoreWLAN.framework/Versions/A/CoreWLAN (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 2487.30.104)
	/System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices (compatibility version 1.0.0, current version 64.0.0)
	/System/Library/Frameworks/CoreGraphics.framework/Versions/A/CoreGraphics (compatibility version 64.0.0, current version 1774.2.3)
	/System/Library/Frameworks/CoreServices.framework/Versions/A/CoreServices (compatibility version 1.0.0, current version 1226.0.0)
	/System/Library/Frameworks/CoreText.framework/Versions/A/CoreText (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 2202.0.0)
	/System/Library/Frameworks/ImageIO.framework/Versions/A/ImageIO (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
```

supermerill, 2025-03-27:
ok, I think I understand the issue.
I'll try to fix it with next nightly, but I may need some iterations.

supermerill, 2025-03-28:
can you test it?
https://github.com/supermerill/SuperSlicer/actions/runs/14126647700/artifacts/2838164332
It seems okay, from the logs.

julian-weinert, 2025-03-29:
@supermerill closer. The Framework related issues are gone, but it's still failing to load `libzstd`:
```
$ SuperSlicer.app/Contents/MacOS/SuperSlicer
dyld: Library not loaded: @executable_path/../Frameworks/libzstd.1.5.7.dylib
  Referenced from: SuperSlicer.app/Contents/MacOS/SuperSlicer
  Reason: no suitable image found.  Did find:
	SuperSlicer.app/Contents/MacOS/../Frameworks/libzstd.1.5.7.dylib: cannot load 'libzstd.1.5.7.dylib' (load command 0x80000034 is unknown)
	SuperSlicer.app/Contents/MacOS/../Frameworks/libzstd.1.5.7.dylib: stat() failed with errno=1
	SuperSlicer.app/Contents/MacOS/../Frameworks/libzstd.1.5.7.dylib: cannot load 'libzstd.1.5.7.dylib' (load command 0x80000034 is unknown)
	SuperSlicer.app/Contents/MacOS/../Frameworks/libzstd.1.5.7.dylib: stat() failed with errno=1
Abort trap: 6
```

otool reports the following:
```
$ otool -L SuperSlicer.app/Contents/MacOS/SuperSlicer
SuperSlicer.app/Contents/MacOS/SuperSlicer:
	/usr/lib/libiconv.2.dylib (compatibility version 7.0.0, current version 7.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 2202.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1600.157.0)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libbz2.1.0.dylib (compatibility version 1.0.0, current version 1.0.8)
	/usr/lib/liblzma.5.dylib (compatibility version 6.0.0, current version 6.3.0)
	@executable_path/../Frameworks/libzstd.1.5.7.dylib (compatibility version 1.0.0, current version 1.5.7)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1336.61.1)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.12)
	/usr/lib/libcurl.4.dylib (compatibility version 7.0.0, current version 9.0.0)
	/System/Library/Frameworks/AudioToolbox.framework/Versions/A/AudioToolbox (compatibility version 1.0.0, current version 1000.0.0)
	/System/Library/Frameworks/WebKit.framework/Versions/A/WebKit (compatibility version 1.0.0, current version 617.1.17)
	/System/Library/Frameworks/Security.framework/Versions/A/Security (compatibility version 1.0.0, current version 61040.61.1)
	/System/Library/Frameworks/Carbon.framework/Versions/A/Carbon (compatibility version 2.0.0, current version 170.0.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 24.0.0)
	/System/Library/Frameworks/QuartzCore.framework/Versions/A/QuartzCore (compatibility version 1.2.0, current version 1.11.0)
	/System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/CoreWLAN.framework/Versions/A/CoreWLAN (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 2487.30.104)
	/System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices (compatibility version 1.0.0, current version 64.0.0)
	/System/Library/Frameworks/CoreGraphics.framework/Versions/A/CoreGraphics (compatibility version 64.0.0, current version 1774.2.3)
	/System/Library/Frameworks/CoreServices.framework/Versions/A/CoreServices (compatibility version 1.0.0, current version 1226.0.0)
	/System/Library/Frameworks/CoreText.framework/Versions/A/CoreText (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 2202.0.0)
	/System/Library/Frameworks/ImageIO.framework/Versions/A/ImageIO (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)

supermerill, 2025-03-31:
Seems like trying to build a 10.14 build on a 10.13 os with a 10.15 deps creates some library version shenanigans.

Can you try the latest nightly? https://github.com/supermerill/SuperSlicer/actions/runs/14177291570/artifacts/2853300798
here it depends on `/usr/local/opt/zstd/lib/libzstd.1.dylib`
I guess you have it, as you said you have something in /usr/local?
What is your macos version btw?

rraallvv, 2025-05-25:
I'm trying to launch the SuperSlicer 2.7.61.3 prerelease, but I keep getting this error: "You have macOS 12.7.5. The application requires macOS 14.2 or later."

Does anyone know if there's any way to get SuperSlicer running on macOS Monterey by building it from source? Hoping for some good news!

supermerill, 2025-05-26:
> Does anyone know if there's any way to get SuperSlicer running on macOS Monterey by building it from source? Hoping for some good news!

You can try to clone the repo, and launch the BuildMacos.sh script (check the args with -h).
You have to have cmake, xcode, zlib installed.
You may need to alter some thigns to make it compile on macOS Monterey.