supermerill/SuperSlicer #4551: Windows: 2.7 Alpha not starting from a link

dl6lr, 2025-03-25:
### What happened?

Superslicer starts fine when started from command line. When trying to start it from a link or by double clicking it, the spinner appears for a fraction of a second and nothing happens.

### Project file & How to reproduce

Use release ZIP and unpack it to any folder.
1. Navigate with Explorer to the folder.
2. Double click superslicer.exe (Using "run as admin" does not work either)
3. Nothing happens
4. Open cmd.exe and cd to the folder
5. Enter "superslicer.exe" and hit the Enter key
6. Superslicer starts with the splash screen

### Version

2.7.61.0_win64_250325

### Operating system

Windows 10

### Printer model

Does not apply

jbeima, 2025-03-25:
Is it being blocked for not being signed by chance?

dl6lr, 2025-03-25:
I don't know, it is not complaining about being unsigned. It is copied to an unsecured folder (not c:\program files\...). Is 2.5.60.0 signed? This (and all previous releases) works as expected on my windows.