supermerill/SuperSlicer #4552: https://superslicer.org not working

roykrikke, 2025-03-25:
### What happened?

Url https://superslicer.org/ is not working.

### Project file & How to reproduce

Open browser and go to https://superslicer.org/

Date: 25-03-2025 17:00 ETC

supermerill, 2025-03-25:
on my browsers (tested with firefox and edge on windows 10), it redirects correctly to https://www.superslicer.org/
not on yours?

roykrikke, 2025-03-25:
Nope I'm sorry. I get:
This site can’t be reached
Check if there is a typo in www.superslicer.org.
DNS_PROBE_FINISHED_NXDOMAIN

supermerill, 2025-03-25:
https://who.is/whois/superslicer.org

seems like your dns don't see what other dns can. Weird.

jbeima, 2025-03-25:
What country are you in? Can you send me the IP of the machine you are using to try and connect to us?

Skaifer, 2025-03-26:
The webpage is accessible only from certain DNS. Public Cloudflare, NextDNS does not work. Quad9, OpenDNS, Google do work. Switched between them with Brave browser.

mrx23dot, 2026-10-28:
Works for me, although!
Error Code: SEC_ERROR_EXPIRED_CERTIFICATE