supermerill/SuperSlicer #4553: V2.7.61.0 - Loading of a model file failed

GP3DS, 2025-03-25:
### What happened?

En enregistrant un projet vide afin de garder en mémoire les couleurs disponibles sur ma machine (ERCF), à l'ouverture de ce projet vide, une erreur apparait et ne charge ni le profil de machine, ni les couleurs, ni le profil d'impression

[image]

### Project file & How to reproduce

[Lightning.3mf.zip](https://github.com/user-attachments/files/19452513/Lightning.3mf.zip)

### Version

2.7.61.0

### Operating system

MacOS 15.3.2

### Printer model

Voron 2.4

supermerill, 2025-03-25:
your 3mf has that in the .rel fie:
```
<?xml version="1.0" encoding="UTF-8"?>
<Relationships xmlns="http://schemas.openxmlformats.org/package/2006/relationships">
 <Relationship Target="/3D/3dmodel.model" Id="rel-1" Type="http://schemas.microsoft.com/3dmanufacturing/2013/01/3dmodel"/>
 <Relationship Target="/Metadata/thumbnail.png" Id="rel-2" Type="http://schemas.openxmlformats.org/package/2006/relationships/metadata/thumbnail"/>
</Relationships>
```
it says that you should have a `/3D/3dmodel.model`
but in your 3mf, you have no 3D directory, and no 3dmodel.model file, hence why it's unloadable.

supermerill, 2025-03-25:
okay, the problem is while saving, it doesn't put an empty model like prusaslicer.

supermerill, 2025-03-25:
fixed for next nightly.
it will be able to write project with correct empty model.
It will also be able to open project with missing model.

supermerill, 2025-03-26:
you can get it here
https://github.com/supermerill/SuperSlicer/actions
then select your macos version, and you should have the link here 

[image]