supermerill/SuperSlicer #4554: Pressure advance calibration, excess zeros and can't define test range small enough.

strayr, 2025-03-25:
### What happened?

[image]

So clockwork (or similar) extruders and 0.4 HF hotends of varying types. Need to test between 0.02 and 0.06, there's no point me going from 0 to 0.1, the test just gets ripped apart at either end of this range.

There's 4 more zeros of precision here than I can step with here at 0.01, if I step at 0.0025 then it's still two to many zeros. It's a mess and impossible to read. r/klippers and similar are full of users transposing a decimal place in their pressure advance, let's not add to the confusion.

### Project file & How to reproduce

[Pressure calibration.zip](https://github.com/user-attachments/files/19458403/Pressure.calibration.zip)

Just run a pressure calibration with the default numbers.

### Version

2.7.61

### Operating system

win 11

### Printer model

Switchwire 373

strayr, 2025-03-26:
> Hi.
> I'm the one who created the tool.
>
> I want to confirm that you're aware you can manually type in values for
> the start/end/ increment values?
>
> The extra 0's are on the to-do list, and will be fixed with the next
> release.

I hadn't seen that.

The layout of modal windows is a bit squished looking on my thinkpad, it's
like 3000ish pixels and running at 250% scale. Was going to check on
something less extra before even trying document that.

Drop-down select boxes you can also type your own values in feel a bit
counterintuitive, I've not seen them for a long time. I'm not a UI expert
by any means and my studies are way out of date, but I think asking a user
to choose from a list or type a value sets clearer expectations of the
interface?

Like I didn't set out to break it or find ways to misuse it, so I wasn't
trying to do that. I'd fixed a printer up that hadn't run for 2 years and
figured I had a SuperSlicer profile that was close enough stashed
somewhere, whereas I'd have to go fishing for a prusaslicer profile and
it's never seen orca. I think most users will default to complying with the
available choices

legend069, 2025-03-26:
The main setup window says **"You can manually edit these values if you want larger or smaller values"** in bold text.
It's not my fault you didn't read it..

The main setup window for the tool is x1600,y600 and can be resized by click dragging on the edges like all windows..
On a 1080p monitor it will take up a significant portion of the window.

Not all users are familiar with how to tune pressure advance / linear advance. So the drop down choices gives them a idea on what values to use, the defaults obviously being a good starting point.

Keep in mind this is the first tool like this that exists, and still needs more work to perfect it.
Going by your picture there is *"gap fill** this is bad, I will have to inspect your config to see what's broken and add the fix for the next release.

closed 2025-04-07 by supermerill