supermerill/SuperSlicer #4555: Position-Dependent X-Axis Scaling for Bedslinger Printers

VooDisss, 2025-03-26:
**Is your feature request related to a problem? Please describe.**
Bedslinger-type 3D printers, especially those with a belt-driven X-axis like the Ender 3 series, often exhibit dimensional inaccuracies specifically in the X-axis. This manifests as prints being undersized in the X dimension compared to the Y axis and the intended dimensions. The root cause is often attributed to the X-axis belt geometry where the belt path is not perfectly parallel to the X-rail, leading to a "cosine error." This error is position-dependent, being more pronounced towards the edges of the print bed and less so in the center.  Current slicer features like global scaling and skew compensation are inadequate to address this non-linear, position-specific dimensional inaccuracy.  Users seeking accurate prints on these popular printer types are often frustrated by this inherent limitation.

**Describe the solution you'd like**
I would like to request a new feature: **Position-Dependent X-Axis Scaling**. This feature would allow SuperSlicer to compensate for the position-dependent dimensional inaccuracies specifically along the X-axis of bedslinger printers.

**Describe how it would work**
This feature would work by:

1.  **Introducing a new Printer Setting:** In "Printer Settings" (or an "Advanced" tab), add a section for "Position-Dependent X-Axis Scaling."
2.  **X-Axis Segmentation:** Allow users to define points or segments along the X-axis (e.g., Left Edge, Center, Right Edge of Bed).
3.  **Scaling Factor per Segment:** For each defined X-axis segment, enable users to input a specific scaling factor. This factor, slightly above 1.0, would compensate for the undersizing at that X-position.
4.  **Interpolation:** SuperSlicer would then interpolate these scaling factors across the entire X-axis range.
5.  **G-code Adjustment:** When generating G-code, for every X-coordinate movement, SuperSlicer would apply the interpolated scaling factor based on the current X-position. This effectively stretches the X-axis movements in areas where the printer is known to undersize.
6.  **Calibration Guidance:** Ideally, SuperSlicer would provide documentation or a guided calibration procedure. This would involve printing a calibration object spanning the X-axis, measuring it at different X-positions, calculating errors, and then inputting the corresponding scaling factors into the settings.

**Describe alternatives you've considered**
Alternatives considered include:

*   **Global Scaling:** This is already available in slicers, but it applies a uniform scaling factor across all dimensions and doesn't address the position-dependent nature of the X-axis error.
*   **Skew Compensation:** Addresses axis misalignment, but not the scaling issue.
*   **Mechanical Fixes:** While mechanically improving the belt path or upgrading components is the ideal long-term solution, it can be complex, require hardware modifications, and might not be feasible or accessible for all users, especially on budget printers. This feature would provide a software-based workaround.

**Additional context**
This feature aims to improve the dimensional accuracy of prints on bedslinger printers, specifically addressing the X-axis inaccuracy issue commonly discussed within the 3D printing community.  It provides a software-based compensation for a known hardware limitation, making it easier for users to achieve more accurate prints without requiring complex mechanical modifications.  This feature would be particularly beneficial for users of popular bedslinger printers like the Ender 3 series who prioritize dimensional accuracy.

This feature request is based on discussions in the 3D printing community, highlighting the problem of X-axis dimensional inaccuracy in bedslinger printers. You can find examples of these discussions here:

*   [https://klipper.discourse.group/t/dimensional-accuracy-problem/2370](https://klipper.discourse.group/t/dimensional-accuracy-problem/2370)
*   [https://klipper.discourse.group/t/correct-dimensional-accuracy/6093](https://klipper.discourse.group/t/correct-dimensional-accuracy/6093)

Implementing this "Position-Dependent X-Axis Scaling" in SuperSlicer would be a valuable advanced feature, enhancing its capabilities for users seeking high dimensional accuracy on their bedslinger 3D printers.

jbeima, 2025-03-26:
Since you are running Klipper all you do to correct for this, is print a test print like Vector 3Ds CaliFlower. Then take the output from it and enter it into the [skew_correction] block for Klipper. Wa la fixed.. 

https://www.klipper3d.org/Skew_Correction.html

jbeima, 2025-03-26:
This should be done at the firmware level and not in the slicer... :)

VooDisss, 2025-03-26:
> Since you are running Klipper all you do to correct for this, is print a test print like Vector 3Ds CaliFlower. Then take the output from it and enter it into the [skew_correction] block for Klipper. Wa la fixed..
> 
> https://www.klipper3d.org/Skew_Correction.html

I tried skew correction - it does not work for this issue. I tried adjusting rotate_distance and skew correction in klipper many times (about 30 times printing a calibrating print which has califlower-like excel calculator and print generator). As mentioned in the forum links the skew and rotation_distance (of stepper motors) value adjustment does not lead to precision.

richfelker, 2025-03-26:
If doing it, I second @jbeima's comment that it should be done in firmware not slicer. Klipper has a very easy framework for adding modules that transform the coordinate system of moves.

But really it should just be done as a physical fix. Shims to fix the belt paths on the Ender 3 are at worst like a 10 minute print.