supermerill/SuperSlicer #4556: Build completely broken from numerous missing #include directives

richfelker, 2025-03-26:
I'm trying to build SuperSlicer (rc branch) from source, and the build keeps stopping with "X does not name a type" errors where the header declaring the type in question clearly is not included from anywhere. So far I've hit at least:

- `<regex>` missing from `LabelObjects.cpp`
- `../BoundingBox.hpp` missing from `LabelObjects.hpp`
- `../Config.hpp` missing from `WipeTower.hpp`
- `../Line.hpp` missing from `Circle.hpp`
- `<tbb/parallel_for.h>` missing from `Thread.hpp`

I can submit a PR with all of them when I finally get it to build (each one requires starting over because of false dependencies forcing everything to get rebuilt) but I'm completely baffled at how this happened and how upstream is building at all. A couple of these might be differences in versions of system libraries, but most are project-internal headers, and I see no way they could ever have worked.

jbeima, 2025-03-26:
We are aware of this. :)

richfelker, 2025-03-26:
OK, but like, how did that happen, and how are you building it? Is there some changeset not checked in to git that has it all fixed?

richfelker, 2025-03-26:
I could see *external* includes being a difference between my system and the Ubuntu build system you're using - things like wxWidgets version, boost version, tbb version, etc.

But the majority of the missing includes are *project-internal* and I don't see any way they could possibly builld successfully, regardless of the specifics of the system you're building them on.

I'll follow up with a complete patch once the build finishes.

supermerill, 2025-03-26:
what's the toolchain cmake created for you?

richfelker, 2025-03-26:
Patch for all the missing includes I hit.  The wx ones are almost surely a matter of me having a newer version that split out some interfaces into their own files, but the rest are clear errors.

```
diff --git a/src/libslic3r/ExtrusionRole.hpp b/src/libslic3r/ExtrusionRole.hpp
index 20ab5c1c55..e845dd7994 100644
--- a/src/libslic3r/ExtrusionRole.hpp
+++ b/src/libslic3r/ExtrusionRole.hpp
@@ -11,6 +11,7 @@
 #include <string>
 #include <string_view>
 #include <cstdint>
+#include <cassert>
 
 namespace Slic3r {
 //that's good and clean but a pain in the ass to debug with the debuggeur.
diff --git a/src/libslic3r/GCode/GCodeFormatter.hpp b/src/libslic3r/GCode/GCodeFormatter.hpp
index cd0c67ba15..2dde5eb68e 100644
--- a/src/libslic3r/GCode/GCodeFormatter.hpp
+++ b/src/libslic3r/GCode/GCodeFormatter.hpp
@@ -10,6 +10,7 @@
 #define slic3r_GFormatter_hpp_
 
 #include "../libslic3r.h"
+#include "../Point.hpp"
 
 #include <string>
 #include <string_view>
diff --git a/src/libslic3r/GCode/LabelObjects.cpp b/src/libslic3r/GCode/LabelObjects.cpp
index bce82a6185..76191687cc 100644
--- a/src/libslic3r/GCode/LabelObjects.cpp
+++ b/src/libslic3r/GCode/LabelObjects.cpp
@@ -7,6 +7,7 @@
 #include "libslic3r/TriangleMeshSlicer.hpp"
 
 #include <boost/regex.hpp>
+#include <regex>
 
 namespace Slic3r::GCode {
 
diff --git a/src/libslic3r/GCode/LabelObjects.hpp b/src/libslic3r/GCode/LabelObjects.hpp
index bb4070dbad..60e1e19349 100644
--- a/src/libslic3r/GCode/LabelObjects.hpp
+++ b/src/libslic3r/GCode/LabelObjects.hpp
@@ -4,6 +4,8 @@
 #include <string>
 #include <unordered_map>
 
+#include "../BoundingBox.hpp"
+
 namespace Slic3r {
 
 enum GCodeFlavor : unsigned char;
diff --git a/src/libslic3r/GCode/WipeTower.hpp b/src/libslic3r/GCode/WipeTower.hpp
index 9a1e4ae1f2..e77aaa3c61 100644
--- a/src/libslic3r/GCode/WipeTower.hpp
+++ b/src/libslic3r/GCode/WipeTower.hpp
@@ -13,6 +13,7 @@
 #include <utility>
 
 #include "libslic3r/Point.hpp"
+#include "../Config.hpp"
 
 namespace Slic3r
 {
diff --git a/src/libslic3r/Geometry/Circle.hpp b/src/libslic3r/Geometry/Circle.hpp
index a83081d157..007d149092 100644
--- a/src/libslic3r/Geometry/Circle.hpp
+++ b/src/libslic3r/Geometry/Circle.hpp
@@ -6,6 +6,7 @@
 #define slic3r_Geometry_Circle_hpp_
[... 145 lines cut ...]
+#include <cassert>
+
 namespace Slic3r { namespace GUI { namespace Widget {
 static int clr_border_hovered = 0x000000; //0xED6B21; // 0x00AE42;
 int clr_background_focused = 0x000000; //0xED6B21;       // 0xEDFAF2;
diff --git a/src/slic3r/GUI/wxExtensions.hpp b/src/slic3r/GUI/wxExtensions.hpp
index d390858685..e5242e1cc0 100644
--- a/src/slic3r/GUI/wxExtensions.hpp
+++ b/src/slic3r/GUI/wxExtensions.hpp
@@ -12,12 +12,14 @@
 #include <wx/combo.h>
 #include <wx/dataview.h>
 #include <wx/button.h>
+#include <wx/bmpbuttn.h>
 #include <wx/sizer.h>
 #include <wx/menu.h>
 #include <wx/bmpcbox.h>
 #include <wx/bmpbndl.h>
 #include <wx/statbmp.h>
 #include <wx/timer.h>
+#include <wx/dcmemory.h>
 
 #include <vector>
 #include <functional>
```

supermerill, 2025-03-28:
Merged the pr, and did a little pass to update some old includes.
The new nightly should be okay, can you test it? (branch `nightly_dev`)

mirlang, 2025-03-30:
> Merged the pr, and did a little pass to update some old includes. The new nightly should be okay, can you test it? (branch `nightly_dev`)

I needed to re-include <math.h> in src/libslic3r/SupportSpotsGenerator.cpp, otherwise builds fine now :)

Please also have a look at what user "Cookie" wrote on Discord (in #general), there are still a few more patches needed to build it against system libraries, on Gentoo or Arch for example

supermerill, 2025-04-01:
> I needed to re-include <math.h> in src/libslic3r/SupportSpotsGenerator.cpp, otherwise builds fine now :)

weird, it has `#include <cmath>`
there shouldn't be anything different in <math.h>
Maybe some lines aren't using `std::` in SupportSpotsGenerator.cpp. If so, can you give me the error log?