supermerill/SuperSlicer #4558: Prusa - Overhang issue

Madraph971, 2025-03-26:
### What happened?

Hello,

When I slice a model, I get overhangs everywhere for no reason on vertical surfaces (100% overlap) and even at 45°

[image]

Tried with other printer (Artillery X1 - Bltouch) nothing comparable.

[image]

Added a Prusa Mk3 for test purpose, same issue.

[image]

I thought it was the model I designed, but on PS 2.9.1 zero overhang as expected.
Tried a torture model ... Hummm something's really wrong
Artillery : 

[image]

Mk4S : 

[image]

[image]

### Project file & How to reproduce

Default profiles and filament

[SS 27610 - overhang issue.zip](https://github.com/user-attachments/files/19476876/SS.27610.-.overhang.issue.zip)

### Version

2.7.61.0

### Operating system

W11 - 24h2

### Printer model

Prusa Mk4S

jbeima, 2025-03-27:
Could you please send us your project(3mf) file please. Thank you!

supermerill, 2025-03-27:
That's an issue I have right now.
You can make them disapear by deactivating dynamic overhangs speed & `overhangs_width_speed`

Madraph971, 2025-03-27:
I deactivated both options, all the unwanted overhangs disappeared.
Also noticed a time calculation error (should be 58min not 1h32, do you want another ticket for it ?

[image]

Thx for the quick reply. And huge thanks for the new version.

supermerill, 2025-03-27:
for the time issue, I can't relly debug with that big of an object.
If you can reproduce it with a 30sec -> 1 min, you can create a new issue.