supermerill/SuperSlicer #4559: 2.7.61.0 does not honour the clearance radius when distributing instances

dl6lr, 2025-03-27:
### What happened?

SuSl 2.7.61.0 shows the clearance radius on the platter, but does not honour it when rearranging objects or filling the bed with instances. 2.5.60.0 still did so

### Project file & How to reproduce

Enable sequential printing
Import an STL to the platter
1) set number of instances
Let SuSl rearrange the instances.
SuSl 2.5.60.0 rearranges all instances with the clearance radius.
SuSl 2.7.61.0 rearranges all instances with minimum distance.
2) Fill bed with instances
SuSl 2.5.60 fills as many instances as will fit
SuSl 2.7.61.0 fills as many instances as would fit without sequential printing

[image]

[image]

[image]

### Version

2.7.61.0

### Operating system

Windows 10 Pro

### Printer model

Ultimaker 2 clone

supermerill, 2025-03-27:
It's complicated.
there is the brim, the skirt, and the skirt can be tall...
After trying to understand & tune it for a week, I gave up.

PizzaScientist, 2025-04-05:
But..... This works perfectly fine in 2.5.59.8. Why change it?

supermerill, 2025-04-14:
Prusa updated it. I merged the new version.
Now, the old way of computing the spacing was very wrong in my tests.

The new very complicated algo allow to rotated the part, and put part inside the hole of another part.

I could have fixed it, but I didn't want to delay everything by a month again.

dl6lr, 2025-04-15:
I may have wrong or too simple use cases, but the previous algo did work for me flawlessly.