supermerill/SuperSlicer #4560: Configuration files

printicus, 2025-03-27:
The latest beta(or alpha, whatever the last one is), after install, prompt me to install configuration files. This had made the last  version that i have instaled(2.5.59.13) to not start anymore. I gues all configuration file go to the same directory. Maibe each version should have its own configuration directory, especially after major changes? What can i do now?

supermerill, 2025-03-27:
What I suggest, for now:
1) in your 2.5.59.13 main directory, create a directory named "configuration"
2) start the 2.5.59.13, it should start as if it's on a new computer. It's to test it works. (just cancel and exit)
3) get into your 2.7 configuration directory (help-> show conf dir)
   * open dir "snapshot"
   * open the lastest one
   * copy-paste the content of this snapshot into your new "configuration" (merge & erase if any)
   * in your new "configuration" rename or erase SuperSlicer.ini
   *  in your new "configuration" rename shapshot.ini into SuperSlicer.ini
now you shouls be able to start 2.5.59 as before, and they won't collide anymore.
I'm planning to overhaul that this year to avoid these kind of issues.

printicus, 2025-03-28:
It works, partially... The v2.5.59.13 starts but not all print setting are available. 

[image]
[image]
[image]

Not a problem. I use the 2.7.

neophyl, 2025-03-28:
Looks like you are in simple mode.  Change that to expert.  Your new 'clean' install will have the default preferences so reset to simple mode.  The configuration profiles dont cover things under the software preferences, you will have to go through those and set them to your liking.

printicus, 2025-03-28:
> Looks like you are in simple mode. Change that to expert. Your new 'clean' install will have the default preferences so reset to simple mode. The configuration profiles dont cover things under the software preferences, you will have to go through those and set them to your liking.

You are right. Din not thinked to look at that. I have not used SS in other than expert mode.

claudioarena, 2025-04-13:
Hello,
I am also getting some issues related to this, I believe.
On the new installation, I do not get any of the filament profiles.

I was able to restore the 2.5 following the directions posted here, but:
- I had to delete one of the filament files, which was crashing the 2.5
- I had to remove the vendor files, which were also crashing the 2.5

I attach here the filament profile that is crashing the 2.5, in case it is useful.
Any suggestion on how to get the filaments to show up on 2.7?

[Generic PLA @CREALITY - custom.zip](https://github.com/user-attachments/files/19726413/Generic.PLA.%40CREALITY.-.custom.zip)

supermerill, 2025-04-14:
> Any suggestion on how to get the filaments to show up on 2.7?
the filament ini files need to be in the "filament" directory of your configuration.
your filament has a condition:
`compatible_printers_condition = printer_notes=~/.*PRINTER_VENDOR_CREALITY.*/`
so it will only qhow up on printers with "PRINTER_VENDOR_CREALITY" in their notes.
Maybe this has disapeared. you can remove it (`compatible_printers_condition =`) so the filament always shows up.