supermerill/SuperSlicer #4565: Bridge Fan Speed possibly not honoured

5teveO, 2025-03-28:
### What happened?

When slicing a small test file I noted that a short bridge, although being identified as being a bridge, was not honoring the bridge fan speed setting - if the dynamic overhang speeds (fan) were enabled however it worked as expected.

I expected when only setting external bridge fan speeds to 100% that the bridge would have the fan running at 100% (although this could possibly be a conflict of settings too)

Bridge clearly showing 'normal' fan speed
[image]

Settings 

[image]

Result with Dynamic overhangs speed (fan) enabled

[image]

### Project file & How to reproduce

[Bridge Failure.zip](https://github.com/user-attachments/files/19496021/Bridge.Failure.zip)

### Version

2.7.61.0

### Operating system

Windows 10

### Printer model

Custom large slinger - but seems to be unaffected by printer

supermerill, 2025-03-28:
set overhangs fan seed?

5teveO, 2025-03-28:
> set overhangs fan seed?

Yes that operates on it correctly but overrides the bridge fan speed completely - however in my humble opinion an overhang is not a bridge.. so if i'm expecting to have a bridge at 100% fan speed and dont want overhangs at the same speed as bridges then I would only set the bridge fan speed

with 100% overhang fan speed result

[image]

legend069, 2025-03-28:
is it detected as a overhang or a bridge?

5teveO, 2025-03-28:
> is it detected as a overhang or a bridge?

Looks like an overhang perimeter - so possibly a setting / thrishold somewhere i'm missing - or any idea why its not seen as a bridge?

[image]

supermerill, 2025-03-28:
bridges are special infill.
overhangs are unsupported perimeters.
they have different constraints and laying patterns.
(currently they can have the same flow, I'll add a separate flow for overhang in next version)

5teveO, 2025-03-28:
> bridges are special infill. overhangs are unsupported perimeters. they have different constraints and laying patterns. (currently they can have the same flow, I'll add a separate flow for overhang in next version)

Ok - so that led me to find and possibly understand the following:-

As the feature is a thin wall and i have 3 perimeters set - SS is finding it as an overhang as all lines of the 'bridge' are in fact perimeters. If i reduce the perimeters to 1 then i get a combination of overhangs and bridges - the perimeters are overhangs and the infill is bridge.

So i guess this is intended behaviour although I find it confusing as i would still consider the overhang perimeters as bridges because they are truly unsupported - in this instance I need the overhang fan speed or dynamic overhang fan speed on to capture this kind of situation.

[image]

supermerill, 2025-03-28:
unsupported periemters are the worst kind of bridges, they are curve, wiggle, and they are still at a "perimeter spacing" from the other ones (ie far far way).

bridges are only strait, be anchored at both sides as much as possible, have their own spacing that let them touch each other (if needed)

If you want only bridges, you can use the `No perimeters on bridge areas`: keep only bridges

5teveO, 2025-03-28:
> unsupported perimeters are the worst kind of "bridges", they are curve, wiggle, and they are still at a "perimeter spacing" from the other ones (ie far far way).
> 
> bridge infill extrusions are only strait, be anchored at both sides as much as possible, have their own spacing that let them touch each other (if needed)
> 
> If you want only bridges, you can use the `No perimeters on bridge areas`: keep only bridges
> 
> I'll rename "bridge" to "bridge infill"
> 
> btw, your wall is not thin enough to be consider as 'thin wall', as you are wider than two perimeters. But l’m nitpicking

I guess my use of 'thin wall' was a bad choise as its an official terminology..  apologies for that :) I appreciate the 'nitpicking'

I will try out the no perimeters on bridge areas and see how that looks - good to know its really just my understanding or lack of... and not an issue.. 

 I think the 'issue' can be closed now.. appreciate the help