supermerill/SuperSlicer #4566: Overhand on object with 2.7.61.0

drewgwallace, 2025-03-30:
### What happened?

When slicing an object today on the latest v2.7.61.0 I found an overhang perimeter that goes across the object a single time on L40.

[image]

I switched from Arachne to Classic and it goes away.

[image]

[image]

### Project file & How to reproduce

[turbo - petg - 0.40n.3mf.zip](https://github.com/user-attachments/files/19527941/turbo.-.petg.-.0.40n.3mf.zip)

Found when slicing on L40. Switching from Arachne to Classic removes artifact.

### Version

2.7.61.0

### Operating system

Win11

### Printer model

Printer for Ants

jbeima, 2025-03-30:
Can you tell me if by chance you have arc generation enabled? If so, could you test it by disabling it please?

drewgwallace, 2025-03-30:
> Can you tell me if by chance you have arc generation enabled? If so, could you test it by disabling it please?

Unless I'm misunderstanding, I do not. I am not doing any post-processing.

jbeima, 2025-03-31:
On the Printer -> General -> Firmware -> G2/G3 Generation / Arc Fitting.

Arc support is build in, has been for a long time you no longer need a post-processing script.

drewgwallace, 2025-03-31:
> On the Printer -> General -> Firmware -> G2/G3 Generation / Arc Fitting.
> 
> Arc support is build in, has been for a long time you no longer need a post-processing script.

I confirmed it shows as disabled.

[image]

supermerill, 2025-04-01:
It's arachne behaving.
Let me check what's going here again.

supermerill, 2025-04-01:
yeah... after investigation, the arachne output botched lines (a point goes to infinite, many missing)
I don't have the resources to learn& debug the internal of arachne. Maybe it will be  patched in 2.9 or maybe someone can fix. it.
Btw, it's quite hard to not make it goes away, any modification and pouf, it's gone.

drewgwallace, 2025-04-01:
> yeah... after investigation, the arachne output botched lines (a point goes to infinite, many missing)
> I don't have the resources to learn& debug the internal of arachne. Maybe it will be  patched [...]

No worries, thanks for all the input!

closed 2025-04-01 by drewgwallace